ARMmbed/mbed-os-example-client #74: Can't compile the example using armcc

javier-moreno-tridonic-com, 2016-08-12:
When I try to compile the example in windows 7 with:
 > \Keil\ARM\ARMCC\bin\armcc --help
Product: MDK-ARM Standard Cortex-M only 5.10
Component: ARM Compiler 5.04 update 1 (build 49)
Tool: armcc [5040049]

Using:
>  mbed compile -m K64F -t ARM

or

> mbed compile -m NUCLEO_F401RE -t ARM

I get in both cases the following output:

Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_CLIENT
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_UVISOR
Scan: FEATURE_IPV4
Scan: FEATURE_IPV6
Scan: FEATURE_STORAGE
Scan: mbed
Scan: env
Compile: BufferedPrint.c
Compile: MyBuffer.cpp
Compile: ESP8266Interface.cpp
[Error] Thread.h@61,0:  #1204: attribute "deprecated" does not take arguments
[Error] Thread.h@86,0:  #1204: attribute "deprecated" does not take arguments
[Error] Thread.h@112,0:  #1204: attribute "deprecated" does not take arguments
[Error] Thread.h@137,0:  #1204: attribute "deprecated" does not take arguments
[Error] RtosTimer.h@47,0:  #1204: attribute "deprecated" does not take arguments
[Error] FunctionPointer.h@32,0:  #1204: attribute "deprecated" does not take arguments
[Error] FunctionPointer.h@37,0:  #1204: attribute "deprecated" does not take arguments
[Error] FunctionPointer.h@49,0:  #1204: attribute "deprecated" does not take arguments
[Error] FunctionPointer.h@54,0:  #1204: attribute "deprecated" does not take arguments
[ERROR] "./mbed-os/rtos/rtos/Thread.h", line 61: Error:  #1204: attribute "deprecated" does not take arguments
"./mbed-os/rtos/rtos/Thread.h", line 86: Error:  #1204: attribute "deprecated" does not take arguments
"./mbed-os/rtos/rtos/Thread.h", line 112: Error:  #1204: attribute "deprecated" does not take arguments
"./mbed-os/rtos/rtos/Thread.h", line 137: Error:  #1204: attribute "deprecated" does not take arguments
"./mbed-os/rtos/rtos/RtosTimer.h", line 47: Error:  #1204: attribute "deprecated" does not take arguments
"./mbed-os/hal/api/FunctionPointer.h", line 32: Error:  #1204: attribute "deprecated" does not take arguments
"./mbed-os/hal/api/FunctionPointer.h", line 37: Error:  #1204: attribute "deprecated" does not take arguments
"./mbed-os/hal/api/FunctionPointer.h", line 49: Error:  #1204: attribute "deprecated" does not take arguments
"./mbed-os/hal/api/FunctionPointer.h", line 54: Error:  #1204: attribute "deprecated" does not take arguments
.\esp8266-driver\ESP8266Interface.cpp: 0 warnings, 9 errors

Digging further it looks like when this part of the code is compiled Is using the wrong definition of **MBED_DEPRECATED** (mbed-os/rtos/rtos/Thread.h):

```C++
MBED_DEPRECATED(
        "Thread-spawning constructors hide errors and may lead to complex "
        "program state when a thread is declared")
    Thread(mbed::Callback<void()> task,
           osPriority priority=osPriorityNormal,
           uint32_t stack_size=DEFAULT_STACK_SIZE,
           unsigned char *stack_pointer=NULL) {
        constructor(task, priority, stack_size, stack_pointer);
    }
```

If I fiddle with the definition:

```C++
#ifndef MBED_DEPRECATED
#if defined(__GNUC__) || defined(__clang__)
#define MBED_DEPRECATED(M) __attribute__((deprecated(M)))
#elif defined(__CC_ARM)
#define MBED_DEPRECATED(M) __attribute__((deprecated))
#else
#define MBED_DEPRECATED(M)
#endif
#endif
```

I can see that **\_\_GNUC__** and **__CC_ARM** are both defined 

I've tested the same procedure in another coworker's machines with the same results.

@MarceloSalazar @markus-becker-tridonic-com

0xc0170, 2016-08-12:
Thanks for reporting. This is a bug in the mbed-os, not with the client. we will send a fix for this (it should check for gnuc and not cc_arm - as we use currentyl --gnu flag for armcc that sets that GNUC macro).

MarceloSalazar, 2016-08-12:
@javier-moreno-tridonic-com please update to MDK v5.20, ARM Compiler 5.06 Update 3
This is the version of the compiler expected to work. Let us know if you continue to see this error.

javier-moreno-tridonic-com, 2016-08-17:
Thanks, updated to MDK v5.20, ARM Compiler 5.06 Update 3 and now it's working

closed 2016-08-17 by javier-moreno-tridonic-com