ARMmbed/mbed-os-example-client #77: incompatible redefinition of macro "UNITY_INCLUDE_CONFIG_H"

maclobdell, 2016-08-22:
from a fresh import of this example 
release tag mbed-os-5.1.2

there are hundreds of warnings for:

[Warning] mbed_config.h@29,0:  #47-D: incompatible redefinition of macro "UNITY_INCLUDE_CONFIG_H" (declared at line 28)

in mbed_config.h, multiple defines are generated for this macro.

// Macros
#define UNITY_INCLUDE_CONFIG_H                                   None                           // defined by library:utest
#define UNITY_INCLUDE_CONFIG_H                                                                  // defined by library:utest

yogpan01, 2016-08-22:
@c1728p9 @0xc0170 Can you please check this ? There are no changes in example code, only change happened is new mbed-os version

sg-, 2016-08-25:
@maclobdell Please file this against mbed-os thx!!

closed 2016-09-02 by yogpan01