ARMmbed/mbed-os-example-client #78: blinking red LED gives impression of failure

BlackstoneEngineering, 2016-08-31:
## Problem
The successful running of the example results in a flashing red LED. This indicated failure to most people.

## Solutions
1. Change the LED deffinition to use led2 which is green on most boards with RGB deffinitions. 
2. Use a blinking LED for failure and a solid LED for success?

BlackstoneEngineering, 2016-09-16:
Please put status_led and led1 on different LED's. It would be even better if you did not use a blinking patten to signify success but rather ran the blinking pattern on failure and a solid LED on success.

BlackstoneEngineering, 2016-09-16:
As it is the constant blinking makes it very hard to demonstrate the client / connector quickstart

anttiylitokola, 2016-11-03:
Changes are now available in master. I'll close this ticket now.

closed 2016-11-03 by anttiylitokola