ARMmbed/mbed-os-example-client #80: A binary built with ARMCC is not able to connect to the connector.

byungdoochoi, 2016-09-08:
**Step:**
1-1. Build this example with the online compilation environment.
1-2. Build this example with mbed cli with -t ARM.
2. Flash the binary and turn on the K64F and ESP8266.
3. See it connects to a wifi AP.

**Symptom:**
It can't connect to the connector; stops moving forward after this line:
\-------
SOCKET_MODE : UDP
Connecting to coap://api.connector.mbed.com:5684
\-------
Enabling logs from ESP8266Interface, I could see below is the last AT command:
\-------
AT> AT+CIPSTART
\-------

**Supposed to be:**
You should see "Registered object successfully!" from your terminal.

yogpan01, 2016-09-12:
@geky Please check this , it works for GCC but not for ARMCC

janjongboom, 2016-09-20:
@yogpan01 @geky Same issue for me, also breaks Simple mbed Client. How can this not be a priority?

yogpan01, 2016-09-20:
@janjongboom which ARMCC build are you using ?

janjongboom, 2016-09-20:
@yogpan01 Online compiler. Load mbed-os-example-client in the online compiler, add security.h, compile -> does not do anything on my K64F...

yogpan01, 2016-11-08:
Hi, there is a new release for mbed OS 5.2 and if there is still an issue with WiFi or some compile version, I suggest to open it up on mbed-os repository as that is the right place to handle such issues, mbed Client only uses those APIs just like any other application and if you are experiencing issues like non-connection on different compiles, it has to be raised on mbed-os.

closed 2016-11-08 by yogpan01