ARMmbed/mbed-os-example-client #83: build failure with Nucleo boards

mridup, 2016-09-16:
compilation fails when built for Nucleo boards. Tried a build with K64F which builds OK.
Adding the ENTROPY macros in mbed_app.json does not add the macros to the mbed_config.h file in the .build/ folder.

`"macros": ["MBEDTLS_USER_CONFIG_FILE=\"mbedtls_mbed_client_config.h\"",
                "MBEDTLS_NO_DEFAULT_ENTROPY_SOURCES",
                "MBEDTLS_TEST_NULL_ENTROPY"],`

In fact the first macro added is repeatedly added to the file multiple times.
E.g. in mbed_config.h (./build/NUCLEO_L476RG/GCC_ARM)
`
#define MBEDTLS_TEST_NULL_ENTROPY
#define MBEDTLS_TEST_NULL_ENTROPY
#define MBEDTLS_TEST_NULL_ENTROPY
#define MBEDTLS_TEST_NULL_ENTROPY
`

error is generated in the file mbed-os/features/mbedtls/inc/mbedtls/check_config.h line no.135
`#error "MBEDTLS_TEST_NULL_ENTROPY defined, but not all prerequisites"`

This failure happens both in mbed-cli and the online compiler.

yogpan01, 2016-09-19:
@mridup Can you please list your mbed-cli and mbed-os versions ?

yogpan01, 2016-09-20:
The latest mbed-os and mbed-cli seems to have different behaviour with config json parameters, this needs to be moved to mbed-os team to verify and propose what has changed. Client can only change documentation when its sure how the mbed Configuration mechanism is working.
@bogdanm @0xc0170 @c1728p9 Please check this and propose what is the correct way to pass these parameters?

yogpan01, 2016-10-26:
Please re-test this with latest mbed OS 5.2 and client example. However, it is strongly recommended NOT to disable security through this work-around.
Please check this https://github.com/ARMmbed/mbed-os-example-client#supported-target-hardware-configurations and also
https://github.com/ARMmbed/mbed-os-example-client#requirements-for-non-k64f-boards again

yogpan01, 2016-11-08:
Hi, We are assuming that this is resolved for you, in case if it is still an issue please reopen this issue or open a new one.

closed 2016-11-08 by yogpan01