ARMmbed/mbed-os-example-client #84: An example of the mbed_app.json file for non-K64F boards is not correct

toyowata, 2016-09-16:
An example of the ```mbed_app.json``` file for non-K64F boards is not correct here:

https://github.com/ARMmbed/mbed-os-example-client#requirements-for-non-k64f-board

This example configuration generates following build error:

```
In file included from ./mbed-os/features/mbedtls/inc/mbedtls/config.h:2585:0,
                 from ./mbed-os/features/FEATURE_CLIENT/mbed-client-mbed-tls/mbed-client-mbedtls/m2mconnectionsecuritypimpl.h:25,
                 from .\mbed-os\features\FEATURE_CLIENT\mbed-client-mbed-tls\source\m2mconnectionsecurity.cpp:22:
./mbed-os/features/mbedtls/inc/mbedtls/check_config.h:135:2: error: #error "MBEDTLS_TEST_NULL_ENTROPY defined, but not all prerequisites"
 #error "MBEDTLS_TEST_NULL_ENTROPY defined, but not all prerequisites"
  ^

[mbed] ERROR: "python" returned error code 1.
```

I think correct configuration for the ```mbed_app.json``` file is something like this:

```
    "target_overrides": {
        "your-target-name": {
            "target.macros_add": ["MBEDTLS_TEST_NULL_ENTROPY", "MBEDTLS_NO_DEFAULT_ENTROPY_SOURCES"]
        },
```

toyowata, 2016-09-16:
Same as https://github.com/ARMmbed/mbed-os-example-client/issues/83

yogpan01, 2016-09-20:
The latest mbed-os and mbed-cli seems to have different behaviour with config json parameters, this needs to be moved to mbed-os team to verify and propose what has changed. Client can only change documentation when its sure how the mbed Configuration mechanism is working.
@bogdanm @0xc0170 @c1728p9  Please check this and propose what is the correct way to pass these parameters?

sg-, 2016-09-22:
@toyowata is correct, macros_add should be used.

yogpan01, 2016-10-26:
@toyowata Can this be closed now ? This has been tested through OOB testing for mbed OS 5.2 and it works fine. Also, we are now asking user to NOT use this feature as it compromises security.

toyowata, 2016-10-26:
@yogpan01 Thanks.

closed 2016-10-26 by toyowata