ARMmbed/mbed-os-example-client #86: problem for sending packet between SED and router

SYuPing, 2016-09-23:
Hi,

I set two endpoints(1 router, 1 sleepy end device) to connect to mDS within gateway, I want to let SED send a packet to router by thread in this example, is this possible and how do I do this?

Many thanks.

yogpan01, 2016-09-23:
@artokin @mikter Can you please provide info for this query ?

mikter, 2016-09-26:
This is not thread issue as thread only gives IP connectivity. 

@hasnainvirk can you help how to use socket API to send message to othre device in internet.

SYuPing, 2016-09-29:
@mikter What you mean is I could use socket(e.x UDP) API content address which is IPv6 preifx + device EUI64 address to finish it?

hasnainvirk, 2016-09-29:
@SYuPing I gave example code with instructions to Sam. Please contact him.

SYuPing, 2016-10-03:
@hasnainvirk Thanks a lot, it is very useful for us.
This problem is gonna be solved and I close this issue.

Thanks.

closed 2016-10-03 by SYuPing