ARMmbed/mbed-os-example-client #93: The device connection problem

SYuPing, 2016-10-06:
Hi,

In my testing environment, there is one router(K64F+6lowpan shield) connected with gateway and registered on mDS. 
I set the led status to blink blue led instead red in main code.(in this case is blinking red led originally)
And after about 3 hours(not exactly, but it would really take a long time), my router will disconnect with both mDS and gateway, also the red led will blink rapidlly but I just set blue led why red led blinky?

So, my problem is
1. Why red led blinky? Whats the meaning of this condition?
2. Is the device just could exist 3 hours or some effective times? Is there any parameter to adjust to keep connection?

Many thanks.

teetak01, 2016-10-06:
Hi @SYuPing 

1+2) Red led blinking can also signal Hard Fault, that would of course drop the device offline

anttiylitokola, 2016-10-07:
Hi, we have made several fixes for reconnection logic which will be part of mbed-os 5.2 release. Please, try again once the release comes public.

jankii01, 2016-10-07:
Hi,

mbedOS development is really open source, so all that work is visible already in GitHub.

https://github.com/ARMmbed/mbed-os/tree/mbed-os-5.2

For a quick look for that. Can you get out any logs via the serial?

teetak01, 2016-10-24:
@SYuPing do you still have issues with the connectivity? Can you test with the latest changes if you are still experiencing problems.

SYuPing, 2016-10-25:
Oh...sorry, I'll try to change my software stack to mbed OS 5.2.
And I close this issue now,
Thanks a lot.

closed 2016-10-25 by SYuPing