ARMmbed/mbed-os-example-client #94: Example is no longer working with 6LoWPAN

stevew817, 2016-10-07:
When doing 'mbed add atmel-rf-driver', it adds the atmel-driver-rf repository on the master branch. It looks like there have been changes there that are not interoperable with the mbed client example causing the build to fail.

Either add a .lib pointing to the right commit to get this to work, or update the readme, please.

yogpan01, 2016-10-10:
@hasnainvirk @SeppoTakalo Please look into this .

teetak01, 2016-10-24:
@stevew817 we have made numerous fixes to the example last week. Also the README has been updated. Your issue should be resolved now. Sorry for the delay.

SeppoTakalo, 2016-11-01:
This should be closed as the original issue has been fixed.

teetak01, 2016-11-07:
Hi @stevenewey I'll close this issue now. Please open a new ticket if you see new issues. Thank you for reporting.

closed 2016-11-07 by teetak01