ARMmbed/mbed-os-example-client #100: Sub-G environment, mesh network could not work, scan fail

meta-johnny, 2016-10-26:
I use this example and keep scan no beacon.

Here is my device
Gateway: FIREFLY 6LOWPAN GATEWAY 900MHZ
Mesh_6lowpan_ND: FRDM-K64F board + mbed 6LoWPAN shield (AT86RF212B)

**I think my Gateway has already up, here is the bootstrap ready log**
====== Gateway log ======
Eth bootstrap ready, IP=fd00:ff1:ce0b:a5e0:fec2:3dff:fe04:81e9
6Lowpan bootstrap ready, IP=fd00:ff1:ce0b:a5e0:0:ff:fe00:31e1
Destination Cache:
Routing table:
                   fe80::/64  if:0 src:'Static' id:0 lifetime:infinite
     On-link (met 128)
 fd00:ff1:ce0b:a5e0:0:ff:fe00:31e1/128 if:0 src:'Loopback' id:0 lifetime:infinite
     On-link (met 128)
 fd00:ff1:ce0b:a5e0:fec2:3dff:fe04:81e9/128 if:1 src:'Loopback' id:0 lifetime:infinite
     On-link (met 128)
                   fe80::/64  if:1 src:'Static' id:0 lifetime:infinite
     On-link (met 128)
     fd00:ff1:ce0b:a5e0::/64  if:1 src:'Static' id:0 lifetime:1791
     On-link (met 64)
RPL memory usage 336
RPL Domain 2000823c
RPL Instance 1
---------------
DODAG fd00:ff1:ce0b:a5e0:0:ff:fe00:31e1
  G=0 MOP=1 Prf=7
    fd00:ff1:ce0b:a5e0::/64  lifetime:infinite pref:-1
                      ::/0   lifetime:infinite pref:0
    fd00:ff1:ce0b:a5e0::/64  lifetime:infinite flags:--*
  Version 240
  *Current version* Rank=0080
DIO trickle Imin=6, Imax=20972, k=10
            I=3072, now=2285, t=2263, c=0
Neighbour Cache 1
Reachable Time: 44577   Retrans Timer: 1000   MTU: 1500
Neighbour Cache 0
Reachable Time: 35539   Retrans Timer: 1000   MTU: 1280
======End Gateway log =======

**And here is my Mesh_6lowpan_ND scan fail log**
====== 6lowpan_ND log ======
[DBG ][m6LND]: Channel: 1
[DBG ][m6LND]: Channel page: 2
[DBG ][m6LND]: Channel mask: 2
[INFO][addr]: Address added to IF 0: fe80::fec2:3d00:4:ab0b
[DBG ][addr]: LL64 Register OK!
[INFO][m6LND]: Start 6LoWPAN ND Bootstrap
[DBG ][6Bo ]: Start Active Scan
[DBG ][mlme]: chan page 2, mask 2
[DBG ][mlme]: MAC: Start MLME scan
[DBG ][mlme]: Scan channel 1
[DBG ][mlme]: BEA REQ tx
[DBG ][mlme]: Scan Complete..Halt MAC
[DBG ][mlme]: Trig confirm
[DBG ][mlme]: Active Scan Result
[DBG ][6Bo ]: Mac scan confirm:No Beacons
[DBG ][mlme]: 00
[DBG ][mlme]: Free Response
[DBG ][6Bo ]: Network Bootsrap Start Fail
[DBG ][mleS]: Free Socket
[DBG ][mlme]: RESET MAC PIB
[DBG ][m6LND]: app_parse_network_event() 3
[DBG ][m6LND]: Link Layer Scan Fail: No Beacons
=====End 6lowpan_ND log==========

and to the mbed_app.json, i have done the follow modified,

        "network-interface":{
            "help": "options are ETHERNET,WIFI,MESH_LOWPAN_ND,MESH_THREAD",
            "value": "MESH_LOWPAN_ND"
        },

            "mbed-mesh-api.6lowpan-nd-channel-page": 2,
            "mbed-mesh-api.6lowpan-nd-channel": 1,
            "mbed-trace.enable": 1

Which is following the sub-G settings,

Do i do anything wrong?

yogpan01, 2016-10-26:
@SeppoTakalo Please look into this.

SeppoTakalo, 2016-10-26:
Hi @meta-johnny 

Where did you get the binary for the SubGhz gateway?
Are you sure that it is the sub-Ghz one?

Can you check does it print out the channel number?

Problem might be that the gateway is running a binary for 2.4 GHz (same driver) but it is just unable to send anything.

meta-johnny, 2016-10-26:
Hi @SeppoTakalo ,

Actually we tired two kinds of method with two devices, one is to use the Gateway Sub-G device and the other one is using Sub-G Shield with K64F board as Border Router.

For K64F board with Sub-G shield, we are pretty sure the channel page and channel is set the same as the K64F Sub-G End Device, since we can get the source code and modify the settings. We printed and make sure the End Device is doing active scan, but the border router did not response.

For Gateway Device, the same as you said, we can not sure the binary also the channel number that flashed to the Gateway Sub-G device.

Thank you

SeppoTakalo, 2016-10-27:
Binaries published in Github are for 2.4 Ghz version of the gateway board, so clearly those are not right ones.

I will look for SubGhz shield and start investigating the K64F border router.

meta-johnny, 2016-10-27:
Hi @SeppoTakalo 

Thank you for the information.
Do you know who should I contact with for the Sub-G gateway board binary? Or at least we can configure to use the right channel.

Thank you

SeppoTakalo, 2016-10-31:
@Metaswd I cannot replicate the problem.

I tested with SubGhz gateway using channels 1 and 10. Both seems to work.

First trial with `mbed-os-example-mesh-minimal`
```
Connecting...
[DBG ][core]: Allocate Root Tasklet
[DBG ][6lo ]: P.Init

[DBG ][pana]: Init
[DBG ][sck ]: Socket id 0 allocated
[DBG ][evlp]: event_loop_thread
[DBG ][core]: NS Root task Init
[DBG ][sck ]: Socket Tasklet Generated
[DBG ][pana]: Pana Tasklet Init
[DBG ][m6La]: init()
[DBG ][m6La]: connect()
[DBG ][sck ]: Socket id 1 allocated
[DBG ][m6LND]: Link-layer security NOT enabled.
[DBG ][m6LND]: Channel: 1
[DBG ][m6LND]: Channel page: 2
[DBG ][m6LND]: Channel mask: 2
[INFO][addr]: Address added to IF 0: fe80::fec2:3d00:4:a903
[DBG ][addr]: LL64 Register OK!
[DBG ][rout]: Added route:
[DBG ][rout]:                    fe80::/64  if:0 src:'Static' id:0 lifetime:infinite
[DBG ][rout]:      On-link (met 128)
[INFO][m6LND]: Start 6LoWPAN ND Bootstrap
[DBG ][mle ]: MLE Tasklet Generated
[DBG ][6Bo ]: Start Active Scan
[DBG ][mlme]: chan page 2, mask 2
[DBG ][mlme]: MAC: Start MLME scan
[DBG ][mlme]: Scan channel 1
[DBG ][mlme]: BEA REQ tx
[DBG ][mMCp]: mac_parse_beacon
[DBG ][BHlr]: Beacon indication: 04:07:4e:65:74:77:6f:72:6b:30:30:30:30:30:30:30:30:30:c1:00
[DBG ][mlme]: Beacon Notify: 04:07:4e:65:74:77:6f:72:6b:30:30:30:30:30:30:30:30:30:c1:00
[DBG ][mlme]: Scan Complete..Halt MAC
[DBG ][mlme]: Trig confirm
[DBG ][mlme]: Active Scan Result
[DBG ][MACh]: LinkQuality: 255, LogicalCh: 1
[DBG ][MACh]: Clean Selected out from the list
[DBG ][MACh]: mac_helper_beacon_payload_reallocate, old len: 0, new: 20
[DBG ][mlme]: MAC: Start network 1 channel 691 panid
[DBG ][6Bo ]: Mac Ready
[DBG ][mlme]: 01
[DBG ][mlme]: Free PAN result
[DBG ][mlme]: Free Response
[DBG ][6Bo ]: Accept & Request
[DBG ][MACh]: Register Device
[DBG ][mMIB]: Over write 0, mac16 ffff mac64: ff:ff:ff:ff:ff:ff:ff:ff, 4294967295
[DBG ][mMIB]: Set 0, mac16 3b99 mac64: fc:c2:3d:00:00:04:d5:07, 0
[DBG ][6Bo ]: RS*
[DBG ][loND]: Create ND version
[DBG ][rout]: Added route:
[DBG ][rout]:  fd00:ff1:ce0b:a5e0:fec2:3d00:4:a903/128 if:0 src:'Loopback' id:0 lifetime:infinite
[DBG ][rout]:      On-link (met 128)
[INFO][addr]: Address added to IF 0: fd00:ff1:ce0b:a5e0:fec2:3d00:4:a903
[DBG ][icmp]: Possible Default Router
[DBG ][rout]: Added route:
[DBG ][rout]:                        ::/0   if:0 src:'RAdv' id:0 lifetime:1024
[DBG ][rout]:      next-hop fe80::ff:fe00:3b99 (met 128)
[DBG ][icmp]: Build NA
[DBG ][loND]: Interface ID: 0, ipv6: fd00:ff1:ce0b:a5e0:fec2:3d00:4:a903
[DBG ][loND]: State Timer CB
[DBG ][loND]: NS Configured
[DBG ][loND]: Interface ID: 0, ipv6: fd00:ff1:ce0b:a5e0:fec2:3d00:4:a903
[INFO][loND]: Address REG OK: fd00:ff1:ce0b:a5e0:fec2:3d00:4:a903
[DBG ][6Bo ]: ND Ready
[... 132 lines cut ...]
[DBG ][6lo ]: priority flag clear all
[DBG ][rplc]: transmit dio, rank: 164
[DBG ][rplc]: Triggering parent selection due to ETX change on fc:c2:3d:00:00:04:d5:07
[DBG ][RPLd]: DAO-ACK RX
[DBG ][mlme]: MAC: Start network 10 channel 691 panid
[DBG ][core]: RPL Ready
[DBG ][m6LND]: app_parse_network_event() 0
[INFO][m6LND]: 6LoWPAN ND bootstrap ready
[DBG ][m6LND]: ND Access Point: fd00:ff1:ce0b:a5e0:0:ff:fe00:c026
[DBG ][m6LND]: ND Prefix 64: fd:00:0f:f1:ce:0b:a5:e0
[DBG ][m6LND]: GP IPv6: fd00:ff1:ce0b:a5e0:fec2:3d00:4:a903
[DBG ][m6LND]: MAC 16-bit: ff:ff
[DBG ][m6LND]: PAN ID: 06:91
[DBG ][m6LND]: MAC 64-bit: fc:c2:3d:00:00:04:a9:03
[DBG ][m6LND]: IID (Based on MAC 64-bit address): fe:c2:3d:00:00:04:a9:03
[DBG ][m6LND]: Channel: 10

Connected to Network successfully
[DBG ][m6LND]: getOwnIpAddress()
IP address fd00:ff1:ce0b:a5e0:fec2:3d00:4:a903
```

If you have SubGhz mbed gateway, you can test with attached binary

[gateway6LoWPANStatic.bin.zip](https://github.com/ARMmbed/mbed-os-example-client/files/561682/gateway6LoWPANStatic.bin.zip)

SeppoTakalo, 2016-10-31:
Also verified that using K64F border router this also works.

I did following changes:
```
diff --git i/mbed_app.json w/mbed_app.json
index 5aab566..b1b5ad2 100644
--- i/mbed_app.json
+++ w/mbed_app.json
@@ -16,7 +16,7 @@
                "model": "FRDM-K64F-BR",
                "manufacturer": "ARM",
                "network-mode": "ND_WITH_MLE",
-               "security-mode": "PSK",
+               "security-mode": "NONE",
                "psk-key-id": 1,
                "psk-key": "{0xa0, 0xa1, 0xa2, 0xa3, 0xa4, 0xa5, 0xa6, 0xa7, 0xa8, 0xa9, 0xaa, 0xab, 0xac, 0xad, 0xae, 0xaf}",
                "pana-mode": "",
@@ -27,11 +27,11 @@
                "beacon-protocol-id": 4,
                "prefix": "fd00:db8::",
                "prefix-from-backhaul": true,
-               "rf-channel": 12,
-               "rf-channel-page": 0,
-               "rf-channel-mask": "0x07fff800",
+               "rf-channel": 10,
+               "rf-channel-page": 2,
+               "rf-channel-mask": "(1<<10)",
                "short-mac-address": "0xface",
-               "backhaul-bootstrap-mode": "NET_IPV6_BOOTSTRAP_AUTONOMOUS",
+               "backhaul-bootstrap-mode": "NET_IPV6_BOOTSTRAP_STATIC",
                "backhaul-prefix": "fd00:db8:ff1::",
                "backhaul-default-route": "::/0",
                "backhaul-next-hop": "fe80::1",
```

**Steps to verify:**

1. Clone K64F border router app.
1. Set BR channel to something between 1 - 10 from `mbed_app.json`. Set also channel mask to correct (see example).
1. Build & flash
1. Clone mesh application, either this or mesh-minimal
1. Set same channel values into `mbed_app.json`
1. Build & flash.
1. Place devices close to each other, just to make sure that they are within their range. I have them on the same table.
1. Start BR. (you can wain 15s or check serial output `6LoWPAN Border Router Bootstrap Complete`)
1. Start application board.
1. Check that client is able to get IPv6 address.

@yogpan01 I think this can be closed, I cannot replicate the problem. All seems to be correct.

Metaswd, 2016-10-31:
@SeppoTakalo 

May I confirm with you again?  Do you mean the binary is for the Gateway device (Atmel MCU +Atmel SubG)? (We did purchase an Gateway device from Firefly)

It's late night here so we will try tomorrow morning. Thank you very much for the testing and helpful information.

SeppoTakalo, 2016-10-31:
Yes. It checked and the binary should work for both 2.4 Ghz and SubGhz.
For subGhz channel 1 is used, for 2.4 Ghz channel 12 is used.

It should use static IPv6 address.

One problem that comes to my mind is that do you have test environment where IPv6 addresses are available in ethernet.
If that is not the case, the dynamic border router does not start. It waits for IPMPv6 Router adverticement messages from Ethernet. The binaries with `-static` suffix will work. They are for use in isolated test environment where IPv6 is not available.

Metaswd, 2016-10-31:
@SeppoTakalo 
Thank you, we will check and hopefully we can make them works. Will feedback here if there is any update.

Metaswd, 2016-11-01:
@SeppoTakalo 

It works now for a K64F+shield connect with Gateway device. We found that we have a bad shield board result in the connection issue, after replace it and follow your step it works now. We will close this issue and thank you again.

closed 2016-11-01 by Metaswd