ARMmbed/mbed-os-example-client #104: Problem of building project

SYuPing, 2016-11-01:
Hi,

I want to update my mbed-os 5.1.2 to 5.2 and I tried to build this project , but it got some trouble, 
the error message is as below. 

C:\Users\04070\WIFI test\mbed-os-example-client-master>mbed compile -m K64F -t G
CC_ARM
Building project mbed-os-example-client-master (K64F, GCC_ARM)
Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_LWIP
Scan: FEATURE_UVISOR
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_NANOSTACK
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_THREAD_ROUTER
Scan: FEATURE_STORAGE
Scan: mbed
Scan: env
Compile [  0.3%]: BusInOut.cpp
[ERROR] arm-none-eabi-g++: error: test/mbed-os-example-client-master/mbed-os/.bu
ild: No such file or directory
arm-none-eabi-g++: error: test/mbed-os-example-client-master/mbed-os/features/un
supported/tests/mbed: No such file or directory
arm-none-eabi-g++: error: test/mbed-os-example-client-master/mbed-os/features/un
supported/tests/mbed/env: No such file or directory

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\Users\04070\WIFI test\mbed-os-example-client
-master\mbed-os\tools\make.py -t GCC_ARM -m K64F --source . --build .\BUILD\K64F
\GCC_ARM" in "C:\Users\04070\WIFI test\mbed-os-example-client-master"
---

How could I solve this?
thx.

jankii01, 2016-11-02:
Hi, 

to which github hash is your mbed-os.lib pointing to?

teetak01, 2016-11-02:
Hi @SYuPing, could you `rm -rf mbed-os `& `mbed deploy` ? That should fix any possible issues in mbed-os folder.

SYuPing, 2016-11-03:
Hi, @jankii01  my mbed-os.lib pointing to 
https://github.com/ARMmbed/mbed-os/#8a9a2463563af3ffb8d289905d3f7073a46ecc80

Hi, @teetak01  I tried....but...it still fail......the result is same as above.....

jankii01, 2016-11-03:
It's not pointing to an official version of mbed OS then. Can you pull in the latest version of this branch, stash your changes 1st.

```
git stash
git pull origin master
git stash pop
(hopefully there are no conflicts)
rm -rf mbed-os
mbed deploy
```

(Better yet if you have no changes in the repo, rm -rf the whole mbed-os-example-client and do a clean git clone). I've noticed that mbed somehow messes up the repos somehow and then it fails with all kinds of strange symptoms.

We have a CI system running this example every day and it's passing.

jankii01, 2016-11-03:
mbed-os 5.2 is https://github.com/ARMmbed/mbed-os/#1d875083b701a5e530f97ba615e2c0d2060d09e5
mbed-os 5.2.1 is https://github.com/ARMmbed/mbed-os/#e435a07d9252f133ea3d9f6c95dfb176f32ab9b6

jankii01, 2016-11-03:
The point where it tries to compile the BusInOut.cpp is very different than what it is for me (I just built it),

```
Compile [  9.4%]: BusIn.cpp
Compile [  9.7%]: BusOut.cpp
Compile [  9.9%]: BusInOut.cpp
Compile [ 10.2%]: Ethernet.cpp
```
while in your log print it was only at 0.3%.

SYuPing, 2016-11-04:
Hi, @jankii01 

The error still exist as I follow the step which you commented...
I copied and paste mbed-os 5.2.1 is https://github.com/ARMmbed/mbed-os/#e435a07d9252f133ea3d9f6c95dfb176f32ab9b6 to my mbed-os.lib

also enter the command as below

> 
git stash
git pull origin master
git stash pop
(hopefully there are no conflicts)
rm -rf mbed-os
mbed deploy

but still got the error...

here is the error message
Compile [  0.3%]: ATParser.cpp
[ERROR] arm-none-eabi-g++: error: test/mbed-os-example-client/mbed-os/.build: No
 such file or directory
arm-none-eabi-g++: error: test/mbed-os-example-client/mbed-os/features/unsupport
ed/tests/mbed: No such file or directory
arm-none-eabi-g++: error: test/mbed-os-example-client/mbed-os/features/unsupport
ed/tests/mbed/env: No such file or directory

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\Users\04070\WIFI test\mbed-os-example-client
\mbed-os\tools\make.py -t GCC_ARM -m K64F --source . --build .\BUILD\K64F\GCC_AR
M" in "C:\Users\04070\WIFI test\mbed-os-example-client"

And now "Compile [  0.3%]: ATParser.cpp" is at 0.3% not BusInOut.cpp .
It's really different....

jankii01, 2016-11-04:
Hi @SYuPing,

have you done any changes? I would feel tempted now to manually copy your latest changes somewhere (git format patch for example, each repo separately). Then clean up the whole client work area, git clone that from clean and re-build. That way you would know if the setup is still working or not. Once you get the basic, vanilla case building completely - start re-applying your changes back (git am).

teetak01, 2016-11-04:
@SYuPing a few things I would like you to check:

- do you have any git-repositories in top level folders, like in: WIFI test, 04070 folders? mbed-cli is pretty sensitive to nested git-repositories and that can cause wierd issues
- can you confirm that you do not have any extra folders in your project? Maybe add the top level folder contents of your project for debugging
- I presume you have been doing clean builds `mbed compile -c`
- reclone clean repository elsewhere and try compiling that

Otherwise if none of the above fix the issue, I would suggest opening a new defect vs.` mbed-cli` and link it here. The above issues are not directly related to the example application.

SYuPing, 2016-11-04:
Hi, @jankii01  , @teetak01 

Yes, I have the same poject name in top level folders, after I delet that, It can work now!!

Maybe mbed-cli compil with two same folders as you said.
So the problem is solved now, I'm very appriciate for your help.
Many thanks.

Close this issue.

closed 2016-11-04 by SYuPing