ARMmbed/mbed-os-example-client #106: Meet SecureConnectionFailed error

ccli8, 2016-11-07:
I've updated `security.h` but still meet the SecureConnectionFailed error. The example worked before.
### Target: K64F
### Toolchain: ARM/GCC_ARM
### Network interface: WiFi
### mbed-os-examle-client version: mbed-os-5.2.1
```
$ mbed ls
mbed-os-example-client (b3655abb7d50)
|- atmel-rf-driver (62c0c49d7a94)
|- esp8266-driver (dc37b65ca877)
|- mbed-client (1c03b36ac944)
|  |- mbed-client-classic (656f13cd4864)
|  `- mbed-client-mbed-tls (7e1b6d815038)
|- mbed-os (e2617cc0e17f)
|- mcr20a-rf-driver (d8810e105d7d)
`- pal (08f7c93f8e24)
```
### VCOM message:
```
Starting mbed Client example...

Using WiFi 

Connecting to WiFi..

Connected to Network successfully
IP address 192.168.43.68

SOCKET_MODE : UDP
Connecting to coap://api.connector.mbed.com:5684

[ERROR:] M2MInterface::SecureConnectionFailed
```

@cyliangtw

anttiylitokola, 2016-11-07:
@geky, can you please check if there is an any issues in WiFi side?

yogpan01, 2016-11-08:
Hi, There seems to be some regression in mbed-os 5.2.1 release, we have reverted the mbed-os to 5.2.0 , please do a git pull on this repo and do `mbed deploy`  to revert back the mbed-os version and verify.

ccli8, 2016-11-08:
Yes, mbed-os 5.2.0 is OK, but mbed-os 5.2.1 is failed. Do you have a plan to merge the fix to master branch of mbed-os?

yogpan01, 2016-11-08:
That needs to be handled by mbed-os. We have raised an internal issue with mbed-os and hopefully they can fix and release it in next version. I am closing this issue for this time though.

closed 2016-11-08 by yogpan01

jankii01, 2016-11-08:
Hi,

mbedOS team will of course fix this issue to some upcoming release.

ccli8, 2016-12-02:
I check with mbed-os-5.3.0-rc1 and the issue is fixed.