ARMmbed/mbed-os-example-client #107: DNS resolution failure in Thread mode

devran01, 2016-11-07:
I've configured the example as a thread end node and it seems like the DNS resolution is failing. I'm getting following error with trace enabled.
```
[DBG ][mClt]: M2MInterfaceImpl::state_register - IP address 2607:f0d0:2601:52::20 , Port 5684
[DBG ][mClt]: resolve_server_address()
[DBG ][mClt]: M2MInterfaceImpl::socket_error: (4), retry (1), reconnecting (0)
```
Digging in deeper, `pal_getAddressInfo()` method in `void M2MConnectionHandlerPimpl::dns_handler()` is failing.

yogpan01, 2016-11-07:
@devran01 Seems that there is a regression introduced with mbed-os-5.2.1 and its not connecting . There is a PR #108 with reverting to previous working mbed-so release.

yogpan01, 2016-11-08:
@devran01 We have reverted the mbed-os 5.2.1 to mbed-os 5.2.0 to resolve the issue, please do the git pull on repository and do `mbed deploy` to revert back to mbed-os 5.2.0

geky, 2016-11-09:
Does https://github.com/ARMmbed/mbed-os/pull/3232 fix this issue?

devran01, 2016-11-15:
@yogpan01 with the commit c57e5ecb74e9d94cd0e3463a3d7ed7872c966324, this issue is not observed anymore and device registers successfully with connector.

@geky I didn't try  https://github.com/ARMmbed/mbed-os/pull/3232

closed 2016-11-15 by devran01