ARMmbed/mbed-os-example-client #112: Client fails to initialise the RF shield for target NUCLEO_F401RE (based in mbed 5.2)

javier-moreno-tridonic-com, 2016-11-11:
The build based on commit c57e5ecb74e9d94cd0e3463a3d7ed7872c966324 not working for target NUCLEO_F401RE and using the GCC_ARM toolchain.

Everytime that I try to execute the resulting binary on my demo board I end up with the following error:

> Connection to Network Failed -3012! Exiting application....

Further digging showed that the it fails to initialise the RF shield since the call to **rf_radio_type_read()** returns **ATMEL_UNKNOW_DEV**

I attached an oscilloscope to the SPI interface and could check that there is activity on the MOSI line (so it manages to write the command to read the radio type) but no activity in the MISO line. 

If attach the same shield with the same probes attached to a K64F, everything works, the radio type is properly read and I can clearly see activity in the MISO line.

@MarceloSalazar @karsev

karsev, 2016-11-11:
@maansaari

yogpan01, 2016-11-11:
@SeppoTakalo Can you please look into this ?

TuomoHautamaki, 2016-11-11:
@sg- , please investigate

sg-, 2016-11-11:
Not supported in https://github.com/ARMmbed/mbed-os-example-client#supported-target-hardware-configurations

Known issue. https://github.com/ARMmbed/oob-mbed-os-example-client/issues/15

javier-moreno-tridonic-com, 2016-11-14:
After reading the pointers and causes commented in ARMmbed/oob-mbed-os-example-client#15 managed to run it by making the following changes to the example:

---
In **simpleclient.h** Replace:
```C++
M2MSecurity* create_register_object() {
        // create security object using the interface factory.
        // this will generate a security ObjectID and ObjectInstance
        M2MSecurity *security = M2MInterfaceFactory::create_security(M2MSecurity::M2MServer);

        // make sure security ObjectID/ObjectInstance was created successfully
        if(security) {
            // Add ResourceID's and values to the security ObjectID/ObjectInstance
            security->set_resource_value(M2MSecurity::M2MServerUri, _server_address);          
            security->set_resource_value(M2MSecurity::SecurityMode, M2MSecurity::Certificate);
            security->set_resource_value(M2MSecurity::ServerPublicKey, SERVER_CERT, sizeof(SERVER_CERT));
            security->set_resource_value(M2MSecurity::PublicKey, CERT, sizeof(CERT));
            security->set_resource_value(M2MSecurity::Secretkey, KEY, sizeof(KEY));            
        }
        return security;
    }
```

With 
```C++
M2MSecurity* create_register_object() {
        // create security object using the interface factory.
        // this will generate a security ObjectID and ObjectInstance
        M2MSecurity *security = M2MInterfaceFactory::create_security(M2MSecurity::M2MServer);

        // make sure security ObjectID/ObjectInstance was created successfully
        if(security) {
            // Add ResourceID's and values to the security ObjectID/ObjectInstance
            security->set_resource_value(M2MSecurity::M2MServerUri, _server_address);
            security->set_resource_value(M2MSecurity::SecurityMode, M2MSecurity::NoSecurity);         
        }
        return security;
    }
```
---
In **mbed_app.json**
 > **"MBEDTLS_NO_DEFAULT_ENTROPY_SOURCES"** and **"MBEDTLS_TEST_NULL_ENTROPY"** need to be added to *macros*

---
In *mbed-os\targets\TARGET_STM\TARGET_STM32F4\TARGET_NUCLEO_F401RE\PinNames.h*

Since PA_5 is used by the SPI clock LEDs need to be "rewired" to an unused pin (e.g. PA_9)

```C++
    LED1        = PA_9,
    LED2        = PA_9,
    LED3        = PA_9,
    LED4        = PA_9,
```

Keep in mind that we register our devices on a Leshan LWM2M server with disabled security for testing purposes. This solution probably won't be useful if the deviced is intended to be used in conjunction with mbed Device Connector

closed 2016-11-14 by javier-moreno-tridonic-com