ARMmbed/mbed-os-example-client #113: Source port randomization

markus-becker-tridonic-com, 2016-11-21:
Looking at
https://github.com/ARMmbed/mbed-os-example-client/blob/master/simpleclient.h#L108
the random source port might be <1024 and thus selecting a registered port. Which might be no problem. At the same time randomly you might choose 5683 (or other locally used ports) by chance. Which might break things. OK, not in this application, but since it is a reference application it might make sense, to fix it here, before it spills in applications by others.

Please implement https://tools.ietf.org/html/rfc6056 with a check_suitable_port() function in this app.

jankii01, 2016-11-22:
Hi @markus-becker-tridonic-com ,

the easiest fix will actually be to change it so that it sets port = 0; 
that way we will use the pretty de-facto way of letting the network stack randomize the port.
However, it seems we'll have to get mbedOS 5.3 release in before we can do that.

jankii01, 2016-12-20:
Hi @markus-becker-tridonic-com,

we have now merged this in: 

https://github.com/ARMmbed/mbed-os-example-client/commit/ece732d633323c8a2412770eba528f260433a93f

- please let me know, if you're OK with this?

JanneKiiskila, 2017-04-19:
Closing out, internal randomizaitons now removed overall and we're using port(0) and letting the stack randomize it.

closed 2017-04-19 by JanneKiiskila