ARMmbed/mbed-os-example-client #125: Socket error with 6lowpan

rdoume, 2016-11-29:
Hey there,
I currently have an issue with the example client, and after trying all i could think, i thought you might have an idea.
My client is one board K64f with the firefly shield in 6lowpan_nd mode, one other K64f in border router mode connected to my pc by ETH, and my LWM2M server being a local leshan server. 
I am able to send regular UDP packet from my client board to my computer, and i'm able to see them through Wireshark, so I know the routing process is correct.
I'm able to connect to the leshan server through the OMA-LWM2M addon for firefox, to see the server.
Here is the trace of my client: 

> **Connected to Network successfully**
> [DBG ][m6LND]: getOwnIpAddress()
> **IP address fd00:db8:ff1:0:fec2:3d00:3:84d2**
> [DBG ][mClt]: M2MInterfaceFactory::create_object : Name : 3200
> [DBG ][mClt]: M2MObject::create_object_instance - id: 0
> [DBG ][mClt]: M2MObject::object_instance(inst_id 0)
> [DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 5501)
> [DBG ][mClt]: M2MResourceInstance::set_value()
> [DBG ][mClt]: M2MResourceInstance::is_value_changed() -- true
> [DBG ][mClt]: M2MResourceInstance::report()
> [DBG ][mClt]: M2MResourceInstance::report() - level 0
> [DBG ][mClt]: M2MResourceInstance::report() - mode = 1, is_observable = 1
> [DBG ][mClt]: M2MInterfaceFactory::create_object : Name : 3201
> [DBG ][mClt]: M2MObject::create_object_instance - id: 0
> [DBG ][mClt]: M2MObject::object_instance(inst_id 0)
> [DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 5853)
> [DBG ][mClt]: M2MResourceInstance::set_value()
> [DBG ][mClt]: M2MResourceInstance::is_value_changed() -- true
> [DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 5850)
> [DBG ][mClt]: M2MInterfaceFactory::create_object : Name : 1000
> [DBG ][mClt]: M2MObject::create_object_instance - id: 0
> [DBG ][mClt]: M2MObject::object_instance(inst_id 0)
> [DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 1)
> [DBG ][mClt]: M2MResourceInstance::set_value()
> [DBG ][mClt]: M2MResourceInstance::is_value_changed() -- true
> [DBG ][mClt]: M2MInterfaceFactory::create_interface - IN
> [DBG ][mClt]: M2MInterfaceFactory::create_interface - parameters endpoint name : ENDPOINT_NAME
> [DBG ][mClt]: M2MInterfaceFactory::create_interface - parameters endpoint type : test
> [DBG ][mClt]: M2MInterfaceFactory::create_interface - parameters life time(in secs): 100
> [DBG ][mClt]: M2MInterfaceFactory::create_interface - parameters Listen Port : 59392
> [DBG ][mClt]: M2MInterfaceFactory::create_interface - parameters Binding Mode : 1
> [DBG ][mClt]: M2MInterfaceFactory::create_interface - parameters NetworkStack : 4
> [DBG ][mClt]: M2MInterfaceFactory::create_interface - Creating M2MInterfaceImpl
> [DBG ][mClt]: M2MNsdlInterface::M2MNsdlInterface()
> [DBG ][mClt]: M2MObject::create_object_instance - id: 0
> [DBG ][mClt]: M2MObject::object_instance(inst_id 0)
> [DBG ][mClt]: M2MObject::object_instance(inst_id 0)
> [DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 0)
> [DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 1)
> [DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 6)
> [DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 7)
> [DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 8)
> [DBG ][mClt]: M2MNsdlInterface::initialize()
> [DBG ][mClt]: M2MInterfaceImpl::M2MInterfaceImpl() -IN
> [DBG ][mClt]: M2MNsdlInterface::create_endpoint( name ENDPOINT_NAME type test lifetime 100, domain DOMAIN, mode 1)
> [DBG ][mClt]: M2MInterfaceImpl::M2MInterfaceImpl() -OUT
> [DBG ][mClt]: M2MInterfaceFactory::create_interface - OUT
> 
>**SOCKET_MODE : UDP**
> **Connecting to coap://[fd00:db8:ff1:0:79df:42e8:293b:9e13]:5683**
> [DBG ][mClt]: M2MInterfaceImpl::set_platform_network_handler()
> [DBG ][mClt]: set_platform_network_handler
> [DBG ][mClt]: M2MInterfaceFactory::create_security
> [DBG ][mClt]: M2MObject::create_object_instance - id: 0
> [DBG ][mClt]: M2MObject::object_instance(inst_id 0)
> [DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 0)
> [DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 1)
> [DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 2)
> [DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 3)
> [DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 4)
> [DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 5)
> [DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 10)
> [DBG ][mClt]: M2MResourceInstance::set_value()
[... 288 lines cut ...]
> [DBG ][mClt]: M2MReportHandler::stop_timers()
> [DBG ][mClt]: M2MReportHandler::stop_timers() - out
> [DBG ][mClt]: M2MReportHandler::set_default_values
> [DBG ][mClt]: M2MNsdlInterface::create_nsdl_resource - Creating in NSDL-C result 0
> [DBG ][mClt]: M2MBase::set_under_observation - observed: 0
> [DBG ][mClt]: M2MBase::set_under_observation - base_type: 1
> [DBG ][mClt]: M2MReportHandler::set_under_observation(observed 0)
> [DBG ][mClt]: M2MReportHandler::stop_timers()
> [DBG ][mClt]: M2MReportHandler::stop_timers() - out
> [DBG ][mClt]: M2MReportHandler::set_default_values
> [DBG ][mClt]: M2MInterfaceImpl::state_register - create_nsdl_list_structure - success
**> [DBG ][mClt]: M2MInterfaceImpl::state_register - server_address coap://[fd00:db8:ff1:0:79df:42e8:293b:9e13]:5683
> [DBG ][mClt]: M2MInterfaceImpl::state_register - IP address fd00:db8:ff1:0:79df:42e8:293b:9e13 , Port 5683**
> [DBG ][mClt]: resolve_server_address()
**> [DBG ][mClt]: M2MInterfaceImpl::socket_error: nterfaceImpl::socket_error: (4), retry (1(4), retry (1), reconnecting (0)**
> [DBG ][mClt]: M2MInterfaceImpl::internal_event : new state 0
> [DBG ][mClt]: M2MInterfaceImpl::state_engine
> [DBG ][mClt]: M2MInterfaceImpl::state_idle
> [DBG ][mClt]: M2MNsdlInterface::stop_timers()
> [DBG ][mClt]: stop_listening()
> [DBG ][mClt]: M2MInterfaceImpl::socket_error - reconnecting in 15(s), count 1/3

I can see that there is a socket error, number 4, although could not find properly what it means.
However, I don't get why there is this error. Maybe you understand? 
Sincerely,

jankii01, 2016-12-01:
Hei,

It's using UDP mode - which network interface are you using? There are many firewalls that have issues with UDP and they close the UDP "hole" after certain time. It is possible you're hitting this issue. If you can, could you try with TCP? Please see section "Changing socket type" in the README.md.

rdoume, 2016-12-01:
The interface would be an eth interface between my 6lowpan border router (https://github.com/ARMmbed/k64f-border-router) and my computer. My client is a node in the 6lowpan network, and therefore uses the AT86RF233, and is connected to the border router and the backhaul.

The TCP mode does not change a thing.
I do not think it's due to my firewall, as, like I said, I can send regular UDP packet that I create  using the UDPsockets and its fonction sendto(); I am able to see in Wireshark those packets going in my interface and use them.
However, while using this example, I cannot even see the packets  going through the interface on wireshark. And this is probably caused by the socket error that you can see in the trace. But I can't explain why this error is happening.

teetak01, 2016-12-01:
@rdoume which version of mbed-client.lib and mbed-os.lib are you using? We recently fixed one issue in client related to WIFI + mbed-os also had some regression with mesh which has been fixed already.

rdoume, 2016-12-01:
My mbed-os.lib is  https://github.com/ARMmbed/mbed-os/#d5de476f74dd4de27012eb74ede078f6330dfc3f
and my mbed-client.lib is : https://github.com/ARMmbed/mbed-client/#1c03b36ac944532688a39d9a6bd7f8a42aecf8f3

teetak01, 2016-12-01:
@SeppoTakalo can you check this? It would seem to failing in DNS-handler.

SeppoTakalo, 2016-12-02:
I tested client on top of LwIP using IPv6 and results are the same:
```
DBG ][mClt]: resolve_server_address()
[DBG ][mClt]: M2MInterfaceImpl::register_object - OUT
[DBG ][mClt]: M2MInterfaceImpl::socket_error: (4), retry (1), reconnecting (0)
[DBG ][mClt]: M2MInterfaceImpl::internal_event : new state 0
[DBG ][mClt]: M2MInterfaceImpl::state_engine
[DBG ][mClt]: M2MInterfaceImpl::state_idle
[DBG ][mClt]: M2MNsdlInterface::stop_timers()
[DBG ][mClt]: stop_listening()
[DBG ][mClt]: M2MInterfaceImpl::socket_error - reconnecting in 15(s), count 1/3
[DBG ][mClt]: M2MInterfaceImpl::timer_expired()
[DBG ][mClt]: M2MInterfaceImpl::internal_event : new state 5
[DBG ][mClt]: M2MInterfaceImpl::state_engine
[DBG ][mClt]: M2MInterfaceImpl::state_register
[DBG ][mClt]: stop_listening()
[DBG ][mClt]: M2MInterfaceImpl::state_register() - new port: 64339
[DBG ][mClt]: resolve_server_address()
```
Using mbed-OS-5.3-RC1

jankii01, 2016-12-05:
Hi, 
would be interesting to get the same results from mbed-OS 5.2.0  (#e435a07d9252f133ea3d9f6c95dfb176f32ab9b6).

jankii01, 2016-12-09:
Hi,

we have been struggling witha bit of a similar case here. The image you compiled for the client (not the border router), what is the mbed_app.json for it like? Seems having too many features turned on there can yield issues like this.

jankii01, 2016-12-14:
Hi @rdoume, 

with ethernet on IPv6 I was able to easily create a similar situation, though I'm not 100% sure if this will apply to you. Please check the mbed_app.json you used for creating the client (not the border router), does it include both "LWIP" and "NANOSTACK"? If it does, please remove the "LWIP".

jankii01, 2016-12-22:
Hi, @rdoume  - any updates on this case?

rdoume, 2016-12-22:
Hi, I'm on leave until beginning January and cannot answer your questions. 
However I managed to make it work somehow, without knowing how. But what you posted about too many features in the mbed_app.json might be it. I'll test is first week of January

jankii01, 2016-12-22:
Thank you @rdoume, excellent to hear you got it working, with your permission I'd like to close this ticket.

JanneKiiskila, 2017-01-05:
Closing this out.

closed 2017-01-05 by JanneKiiskila