ARMmbed/mbed-os-example-client #130: Stack overflow

betzw, 2016-12-02:
When first using the POST and then the PUT methods I get a stack overflow error as follows:
```
simulate button_click, new value of counter is 23

PUT Request Received!
Name :'5853',
Type : '1' (0 for Object, 1 for Resource),
Type : 'Pattern'
simulate button_click, new value of counter is 24
led_execute_callback pattern=75:75:75:75:75:75:75:75:75:75:75:75:75:75:75:75:75:75:75:75:75:75:75:75:75:75:75
Resource: 3201/0/5850 executed
Payload:
RTX error code: 0x00000001, task ID: 0x2001305C
```

teetak01, 2016-12-05:
Thanks for reporting, we'll check this.

jankii01, 2016-12-12:
It's likely having a stack overflow, as it calls itself in a repeat manner with the timer - there's so many entries in the LED pattern.