ARMmbed/mbed-os-example-client #136: Thread SED configured with TCP fails

MarceloSalazar, 2016-12-09:
Tested 5.3-rc3 (https://github.com/ARMmbed/mbed-os-example-client/tree/mbed-os-5.3-rc3)

Default option TCP fails. When changed to UDP works ok (device is registered in mDC).

Platform K64. Atmel RF.

[mbed_app.json.txt](https://github.com/ARMmbed/mbed-os-example-client/files/642975/mbed_app.json.txt)

@jankii01

jankii01, 2016-12-10:
True, mesh only supports UDP.

jankii01, 2016-12-10:
Please see this for example; https://github.com/ARMmbed/mbed-os-example-client/pull/124

jankii01, 2016-12-11:
This fixes it: https://github.com/ARMmbed/mbed-os-example-client/pull/139

jankii01, 2016-12-11:
@MarceloSalazar - please confirm the fix.

MarceloSalazar, 2016-12-11:
@jankii01 thanks for the fix. I'm going to test this tomorrow asap.

In the meantime, could you have a look to the markdown doc?

- IP address setup (https://github.com/ARMmbed/mbed-os-example-client#ip-address-setup)
Ethernet now supports IPv4 and IPv6. How to select this is unknown.

- Changing socket type (https://github.com/ARMmbed/mbed-os-example-client#changing-socket-type)
Need to explain whether a network interface supports UDP, TCP or both. Something like this would help (need to double check whether is correct):

| Network  interface                    | UDP   | TCP | 
| ------------------------------|:-----:|:-----:|
| Ethernet (IPv4)               |   X   |   X   | 
| Ethernet (IPv6)               |   X   |       | 
| Wifi (IPv4)                   |   X   |   X   |
| Wifi (IPv6) - Not supported (?)  |       |       |
| 6LoWPAN/Thread (IPv6)         |   X   |       |

MarceloSalazar, 2016-12-13:
I confirm the original issue is fixed:

Tested latest commit (https://github.com/ARMmbed/mbed-os-example-client/commit/8ce84b520ae0da685bf800addd4dcfc8401acc58) on mbed-OS-5.3 branch.

K64, GCC_ARM, mbed GW
Device successfully registered in mDC.

closed 2016-12-13 by MarceloSalazar