ARMmbed/mbed-os-example-client #137: Add default JSON files to 'config' folder

MarceloSalazar, 2016-12-09:
There are lots of configurations available for this example. Modifying 'mbed_app.json' is becoming tedious.

I suggest to create a 'config' folder and add multiple json files that could be reused by the user (by copying it to the root directory and renaming to 'mbed_app.json'). Suggestions:

- Ethernet_IPv4
- Ethernet_IPv6
- Wifi_IPv4
- Mesh_6LoWPAN_AT233
- Mesh_Thread_SED_AT233
- Mesh_Thread_Router_AT233
- Others?

@jankii01

jankii01, 2016-12-10:
I have to say that would make sense. We have some rather nifty execute sed /s stuff in the Jenkins scripts etc., those could be simplied to simple cp config/xxx.json mbed_app.json -calls.

jankii01, 2016-12-14:
Closing this issue, master has now those.

closed 2016-12-14 by jankii01