ARMmbed/mbed-os-example-client #141: Fail building with uvision5

SYuPing, 2016-12-12:
Hi, 

I want to use Keil to build this project, and I command "mbed export -i uvision5" on mbed-cli, 
Then I open the file which was exported and try to build, but the error shows:

Error: *** Project 'mbed-os-example-client-master', Target 'mbed-os-example-client-master', Device 'Freescale::MK64FN1M0xxx12' is not supported by Toolchain ! ***.
Build Time Elapsed:  00:00:00

How to solve this?
the toolchain which I used is gcc not ARMCC, is this reason caused?

SeppoTakalo, 2016-12-12:
Use Keil pack installer to install the support packages for these boards.

[image]

For me, its in `C:\Keil_V4\UV4\PackInstaller.exe`

Search for `K64F`
On the right panel, install required packs.

SYuPing, 2016-12-14:
OK, I could build project after I installed required packs.
Thanks a lot.

Close this issue

closed 2016-12-14 by SYuPing