ARMmbed/mbed-os-example-client #167: Send data only if the device is successfully registered

MarceloSalazar, 2017-01-23:
We see the application starts sending data periodically to mDC, regardless on whether the device is registered or not.
I believe this doesn't makes sense, as it confused people (especially when the application is not working correctly due to other reasons).

Could you please consider putting the data transmission on hold until the device is successfully registered?

janjongboom, 2017-02-07:
I believe this should be handled by mbed Client.

Writing to a resource before registering should be allowed, but just not yield a call to Connector.

yogpan01, 2017-02-07:
@MarceloSalazar Do you mean the console prints are confusing to the users in case of non-K64F boards?

MarceloSalazar, 2017-02-07:
Correct. Console prints are confusing and shouldn't be displayed until the device is registered (I tried this on a non-working NUCLEO platform and was still seeing printf messages).

janjongboom, 2017-02-07:
I don't agree. It's fine to show the updated state of f.e. a button. As long as the correct value is sent to Cloud when register happens.

It would be confusing if a user presses a button before registering and no message shows.

teetak01, 2017-02-08:
Maybe we could just add check for the `bool registered` and not do simulated button_press if that is false? Is this the issue?

MarceloSalazar, 2017-02-08:
I understand what @janjongboom is saying and believe he's right: when a push button is pressed, a message should be shown.
However, as we are having a simulated counter (for non-k64), maybe we could postpone the execution until 'registered' is true as @teetak01 is suggesting.
These are ideas, although it's not a critical issue.

anttiylitokola, 2017-02-21:
Fix merged in. Closing the issue.

closed 2017-02-21 by anttiylitokola