ARMmbed/mbed-os-example-client #168: Broken: UBLOX_EVK_ODIN_W2 (Ethernet)

andcor02, 2017-01-25:
Importing example and switching to target UBLOX_EVK_ODIN_W2 as Ethernet or WiFi, compile all fine, flashes fine.

**ISSUE**
1. Board just hangs, outputs "Connecting to ethernet..." but nothing else happens?

2. When using WiFi, connects to network and registers, however after one button press on 'SW1' again device hangs.

?

teetak01, 2017-01-25:
Hi, @andcor02 there is some flashing related issues with the UBLOX. If you try flashing the same binary twice (or do it  in windows) it should work better.

andcor02, 2017-01-25:
@teetak01 

Tried both options still the same issue, i think the WIFI issue is also related.

JanneKiiskila, 2017-01-25:
Hi @andcor02,

can you share more details on your project. I have tested the UBLOX in OOB testing and it worked then and well, just tried it out in our RaaS with WiFi and it seems to work. I can't access the HW physically so I can't try what the button press would do - but at least the simulated events go through just fine.

Ethernet I also tried, but I can't be 100% sure now the ethernet is even connected on those HWs, so I can't confirm that.

I.e. share your work area settings - what version of mbedOS you have etc.? 
This result with the current master of the mbed-os-example-client @ b37d883524ab2957238c743958ff7d22616b77ed and everything else via mbed deploy,
mbed_app.json copied from configs/wifi_v4.json -> mbed_app.json and wifi settings injected.

JanneKiiskila, 2017-01-25:
@andcor02 - this;

Board just hangs, outputs "Connecting to ethernet..." but nothing else happens?

-> usually means it can't get an IP address for one reason or another.

JanneKiiskila, 2017-01-26:
With WiFi on - the device has stayed stable whole night, button_click counter is now 2827 and I can read it with REST API etc.

JanneKiiskila, 2017-01-26:
@andcor02  - By default the buttons do not do anything in the client example - so some modifications have been made, can you share those?

andcor02, 2017-01-26:
Hi Janne,

Thanks for all the help.

The Ethernet issue I'm still not sure of, just doesn't seem to get an IP.

And yes i modified the example at the latest revision to allow the interrupt to happen on physical button. 

I have attached my example program here if it helps: https://developer.mbed.org/users/andcor02/code/mbed-os-demo-client-WiFi-notPUBLISHED/

Regards,

Andrea

andcor02, 2017-01-26:
@JanneKiiskila 

I should also add, i have tested this base program above on two other platforms and has worked just fine, the is the only one throwing problems.

JanneKiiskila, 2017-01-26:
@andcor02  -- REAMDE.md updated via https://github.com/ARMmbed/mbed-os-example-client/pull/169, that was erroneus. Not able to get UBLOX ethernet working on this end either, but WiFi works so I'm not seeing an issue with the mbed-os-example-client itself that needs fixing at this stage.

Issues are with UBlox ethernet driver - need to have an mbedOS issue open for that separately, which the UBLOX team needs to get resolved.

andcor02, 2017-01-26:
@JanneKiiskila 

Are you able to toggle SW1 and see it update it in connector?

This is the issue and workflow i have, i can see it update once, but then the whole system seems to hang?

Have you tried my example?

Thanks a lot :)

JanneKiiskila, 2017-01-26:
Haven't tried the example, haven't had the time.

JanneKiiskila, 2017-01-27:
Actually, the ethernet also WOULD work - but ONLY if you know the tricks, there's some EMAC limitation now and the WiFi is the one that works by default.

Quote from:  https://developer.mbed.org/platforms/ublox-EVK-ODIN-W2/

"Only possible to use either Wi-Fi or Ethernet as MAC-interface, not both at the same time. This is a limitation in mbed-os currently. Wi-Fi is enabled by default. To use Ethernet please remove "EMAC" from the UBLOX_EVK_ODIN_W2 target's "device_has" field in mbed-os/targets/targets.json."

JanneKiiskila, 2017-02-01:
Based on my testing, you can't do the overwite via mbed_app.json - it should happen this way;

https://github.com/ARMmbed/mbed-os-example-sockets/blob/master/mbed_app.json#L3

BUT - the only way I got it working was to directly remove the "EMAC" from mbed-os/TARGETS/targets.json file under the UBLOX_EVK_ODIN_W2 entry.

JanneKiiskila, 2017-02-01:
Well, managed to create an mbed_app.json that actually works with UBLOX as well.

JanneKiiskila, 2017-02-01:
https://github.com/ARMmbed/mbed-os-example-client/pull/173 PR is in now.

JanneKiiskila, 2017-02-02:
Closing out, ethernet and WiFi both work in the example client.

closed 2017-02-02 by JanneKiiskila

JanneKiiskila, 2017-02-06:
@andcor02  - please see this issue; https://github.com/ARMmbed/mbed-os/issues/3699
I think the SW0/SW1 do not have the correct mappings in mbedOS.
These mappings do nothing for me, but the client is NOT stuck. It responds just nicely to the PUT/POST requests, so it's running.