ARMmbed/mbed-os-example-client #174: mbed_apps.json: Cannot remove "NANOSTACK" from "target.features_add"

adustm, 2017-02-02:
Hi,
I am trying to use wifi (ESP8266 module) together with a NUCLEO_F429ZI platform.
I modified the mbed_apps.json based on the configs.wifi_v4.json file
```
    "target_overrides": {
        "*": {
            "target.features_add": ["LWIP", "COMMON_PAL"],
            "lwip.ipv4-enabled": true,
            "lwip.ipv6-enabled": false,
            "mbed-trace.enable": 0
        },
```
But then typing `mbed compile -t GCC_ARM -m NUCLEO_F429ZI` does not work. I need to add `"NANOSTACK"` to make the compilation successfull again.
It looks like mbed tries to compile the atmel driver, but as nanostack is not defined, it does not find the arm_hal_phy.h file.

Am I the only one having this trouble ?
Could anyone remove this dependency ? 
Could we avoid compiling the atmel rf driver when not needed ?

See below the details:
```
$ mbed compile -t GCC_ARM -m NUCLEO_F429ZI
Building project mbed-os-example-client (NUCLEO_F429ZI, GCC_ARM)
Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_LWIP
Scan: FEATURE_UVISOR
Scan: FEATURE_ETHERNET_HOST
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_NANOSTACK
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_THREAD_ROUTER
Scan: FEATURE_STORAGE
Scan: mbed
Scan: env
Compile [  0.3%]: NanostackRfPhyAtmel.cpp
[ERROR] .\atmel-rf-driver\source\NanostackRfPhyAtmel.cpp:18:44: fatal error: nanostack/platform/arm_hal_phy.h: No such file or directory
compilation terminated.

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\EXAMPLES\mbed-os-example-client\mbed-os\tools\make.py -t GCC_ARM -m NUCLEO_F429ZI --source . --build .\.build\NUCLEO_F429ZI\GCC_ARM"
in "C:\EXAMPLES\mbed-os-example-client"
```

MarceloSalazar, 2017-02-02:
There is a 'hack' needed as workaround to that build problem. See info here.
https://github.com/ARMmbed/mbed-os-example-client#compilation-problems

We also recommend using the pre-defined json files:
https://github.com/ARMmbed/mbed-os-example-client/tree/master/configs

We apologies for the inconvenience this has caused to you. We'll try to fix this soon.

JanneKiiskila, 2017-05-10:
Documentation was updated to match the need to have the .mbedignore in case you're not using mesh.
Please raise a new ticket to mbed-cli if you think this is not sufficient.

closed 2017-05-10 by JanneKiiskila