ARMmbed/mbed-os-example-client #175: Problem connecting mbed os example client with dtls enabled to Leshan

adriankueng, 2017-02-03:
I´m trying to connect a mbed os client with DTLS to a Leshan. First I used to connect to the ARM connector with their credentials and it worked properly. The next thing I tried was a setup with a local leshan with the PSK ciphersuite, but i could not get it working. So i switched to the Certificate ciphersuite, where I got it running one time, but they only exchanged one package each. Since then I`m stuck on the 6th flight (client sends "ChangeCipherSpec" and "Encrypted Handshake Message") and the server gets back to the 4th flight (```"Server Hello" ... "..Done"```).

```
"No.","Time","Source","Destination","Protocol","Length","Info"
"1","0.000000","Client"----->"Server","DTLSv1.0","181","Client Hello"
"2","0.225764","Server"<-----"Client","DTLSv1.2","122","Hello Verify Request"
"3","0.297524","Client"----->"Server","DTLSv1.0","213","Client Hello"
"4","0.506469","Server"<-----"Client","DTLSv1.2","1225","Server Hello, Certificate"
"5","0.510098","Server"<-----"Client","DTLSv1.2","415","Server Key Exchange, Certificate Request, Server Hello Done"
"6","1.534889","Server"<-----"Client","DTLSv1.2","1225","Server Hello, Certificate"
"7","1.538469","Server"<-----"Client","DTLSv1.2","415","Server Key Exchange, Certificate Request, Server Hello Done"
"8","3.639545","Server"<-----"Client","DTLSv1.2","1225","Server Hello, Certificate"
"9","3.642932","Server"<-----"Client","DTLSv1.2","415","Server Key Exchange, Certificate Request, Server Hello Done"
"10","3.723168","Client"----->"Server","DTLSv1.2","592","Certificate"
"11","6.530284","Client"----->"Server","DTLSv1.2","185","Client Key Exchange"
"12","7.352226","Client"----->"Server","DTLSv1.2","162","Certificate Verify"
"13","7.356961","Client"----->"Server","DTLSv1.2","76","Change Cipher Spec"
"14","7.381765","Client"----->"Server","DTLSv1.2","115","Encrypted Handshake Message"
"15","7.544108","Client"----->"Server","DTLSv1.2","592","Certificate"
"16","7.548869","Client"----->"Server","DTLSv1.2","185","Client Key Exchange"
"17","7.553294","Client"----->"Server","DTLSv1.2","162","Certificate Verify"
"18","7.558359","Client"----->"Server","DTLSv1.2","76","Change Cipher Spec"
"19","7.562992","Client"----->"Server","DTLSv1.2","115","Encrypted Handshake Message"
"20","7.782404","Server"<-----"Client","DTLSv1.2","1225","Server Hello, Certificate"  // !!!!
"21","7.786347","Server"<-----"Client","DTLSv1.2","415","Server Key Exchange, Certificate Request, Server Hello Done"
"22","7.997052","Server"<-----"Client","DTLSv1.2","129","Change Cipher Spec, Encrypted Handshake Message"
"23","10.139419","Client"----->"Server","DTLSv1.2","592","Certificate"
"24","10.144136","Client"----->"Server","DTLSv1.2","185","Client Key Exchange"
"25","10.149045","Client"----->"Server","DTLSv1.2","162","Certificate Verify"
"26","10.153632","Client"----->"Server","DTLSv1.2","76","Change Cipher Spec"
"27","10.158366","Client"----->"Server","DTLSv1.2","115","Encrypted Handshake Message"
```

____
 **My setup:**
 **i added these cipher suites to mbedtls_mbed_client_config.h**

    MBEDTLS_TLS_ECDHE_ECDSA_WITH_AES_256_GCM_SHA384,

    MBEDTLS_TLS_ECDHE_ECDSA_WITH_AES_128_GCM_SHA256,

    MBEDTLS_TLS_ECDHE_ECDSA_WITH_AES_128_CCM_8,

    MBEDTLS_TLS_ECDHE_ECDSA_WITH_AES_128_CBC_SHA256

these should work with the leshan
(http://www.programcreek.com/java-api-examples/index.php?api=org.eclipse.californium.scandium.dtls.cipher.CipherSuite).

 **For the security credentials I used the follwing script: ´create-keystores.sh´**:
```
#!/bin/bash

KEY_STORE=keyStore.jks
KEY_STORE_PWD=endPass
TRUST_STORE=trustStore.jks
TRUST_STORE_PWD=rootPass
VALIDITY=365

echo "creating root key and certificate..."
keytool -genkeypair -alias root -keyalg EC -dname 'C=CA,L=Ottawa,O=Eclipse IoT,OU=Californium,CN=cf-root' \
        -ext BC=ca:true -validity $VALIDITY -keypass $TRUST_STORE_PWD -keystore $TRUST_STORE -storepass $TRUST_STORE_PWD

echo "creating CA key and certificate..."
keytool -genkeypair -alias ca -keyalg EC -dname 'C=CA,L=Ottawa,O=Eclipse IoT,OU=Californium,CN=cf-ca' \
        -ext BC=ca:true -validity $VALIDITY -keypass $TRUST_STORE_PWD -keystore $TRUST_STORE -storepass $TRUST_STORE_PWD
keytool -keystore $TRUST_STORE -storepass $TRUST_STORE_PWD -certreq -alias ca | \
  keytool -keystore $TRUST_STORE -storepass $TRUST_STORE_PWD -alias root -gencert -validity $VALIDITY -ext BC=0 -rfc | \
  keytool -alias ca -importcert -keystore $TRUST_STORE -storepass $TRUST_STORE_PWD

echo "creating server key and certificate..."
keytool -genkeypair -alias server -keyalg EC -dname 'C=CA,L=Ottawa,O=Eclipse IoT,OU=Californium,CN=cf-server' \
        -validity $VALIDITY -keypass $KEY_STORE_PWD -keystore $KEY_STORE -storepass $KEY_STORE_PWD
keytool -keystore $KEY_STORE -storepass endPass -certreq -alias server | \
  keytool -keystore $TRUST_STORE -storepass $TRUST_STORE_PWD -alias ca -gencert -validity $VALIDITY -rfc > server.pem
keytool -alias server -importcert -keystore $KEY_STORE -storepass $KEY_STORE_PWD -trustcacerts -file server.pem

echo "creating client key and certificate..."
keytool -genkeypair -alias client -keyalg EC -dname 'C=CA,L=Ottawa,O=Eclipse IoT,OU=Californium,CN=cf-client' \
        -validity $VALIDITY -keypass $KEY_STORE_PWD -keystore $KEY_STORE -storepass $KEY_STORE_PWD
keytool -keystore $KEY_STORE -storepass $KEY_STORE_PWD -certreq -alias client | \
  keytool -keystore $TRUST_STORE -storepass $TRUST_STORE_PWD -alias ca -gencert -validity $VALIDITY -rfc > client.pem
keytool -alias client -importcert -keystore $KEY_STORE -storepass $KEY_STORE_PWD -trustcacerts -file client.pem
```
 
then modified security.h with following changes:

    const uint8_t SERVER_CERT[] =  ´CA certificate from truststore.jks alias CA´

    const uint8_t CERT[] = ´client certificate from keystore.jsk alias client´

    const uint8_t KEY[] = ´private key from keystore.jsk alias client´

 **I modified LeshanServerDemo.java so that it uses X509 certificates**
(every line with my name behind it was changed)
in words: I changed the path to the keystores, changed the passwords, loaded the certificates and the key, put it in corresponding arrays and add the things to the builder.

___
**Questions:**
* Did anyone test a setup like this?
* Am i using the right credentials in the mbed client?
* Is the loading of the credentials in the server wrong?

---
 **Versions:**
* mbed client SHA-1: a89b5bc90c3d9cb6d17e66a8c2d6e51b8bacdfc9 
* leshan: leshan-server-demo-0.1.11-M15-SNAPSHOT (with  californium 2.0.0 M2)

___

**Attachements:**
* LeshanServerDemo.java: src file of the server creation
[LeshanServerDemo.txt](https://github.com/ARMmbed/mbed-os-example-client/files/750494/LeshanServerDemo.txt)
* loghtermserial.txt: logfile from the serial logger
[loghtermserial.txt](https://github.com/ARMmbed/mbed-os-example-client/files/750496/loghtermserial.txt)
* logleshan.txt: l4j2 loggings from the Leshan server
[logleshan.txt](https://github.com/ARMmbed/mbed-os-example-client/files/750497/logleshan.txt)

* create-keystores.sh: scriptfile to create the credentials
[create-keystores.txt](https://github.com/ARMmbed/mbed-os-example-client/files/750501/create-keystores.txt)

markus-becker-tridonic-com, 2017-02-06:
FYI: @MarceloSalazar @devran01

yogpan01, 2017-02-07:
Hi,
Can you try to pass Server certificate here
```const uint8_t SERVER_CERT[] =  ´ Server certificate from truststore.jks alias server´ ```

adriankueng, 2017-02-08:
I got it working:

I had to add the CN of the client in `mbed-client\security.h` like this:  
*cn = common name; can be found in the certificate*  

    #define MBED_ENDPOINT_NAME "cf-client"

And increase the RetransmissionTimout-Time of the Leshan in 
`leshan\leshan-server-cf\src\main\java\org\eclipse\leshan\server\californium\impl\LeshanServer.jav`

	builder.setRetransmissionTimeout(10000);

Last but no least add this endpoint name to the internal security store of the leshan in 
`\leshan\leshan-server-demo\src\main\java\org\eclipse\leshan\server\demo\LeshanServerDemo.java`

	securityStore.add(SecurityInfo.newX509CertInfo("cf-client"));

@yogpan01 yes, that change alone didn\`t do the trick. But you can use either of them in the `uint8_t SERVER_CERT[]`

one last question:

Am i supposed to add the CertInfo/CommonName/Endpoint in the leshan? I thought this should happen automatically after the handshake is finished. But i guess this is a Leshan problem.

yogpan01, 2017-02-13:
Good that you got this working. I am not sure what you need to do on the Leshan server side.
From client end, once the DTLS handshake is successful, we send the registration message to the server in which we set the endpoint name. So, that's what we do on client and it should appear as registered end point on server side.

yogpan01, 2017-02-16:
Since this is no longer client issue, I am closing this.

closed 2017-02-16 by yogpan01