ARMmbed/mbed-os-example-client #178: problem on reading a resource value with REST API node-red

masoudr, 2017-02-10:
Hi,
I created a simple program with mbed client and mbed os which creates a resource called "/device/0/name" on my device. I can read the resource value with `GET `or `PUT `new value on it with mbed connector dashboard with no problem. I create a node red sample which does the same thing.  If I use this command on my node red it works fine:
`GET https://api.connector.mbed.com/v2/endpoints/<my device id>`
and it response the values which are:
`[{"uri":"/device/0/name","obs":true,"type":""},{"uri":"/3/0","obs":false,"type":""}]"`
and it is OK. The problem is when I want to get a resource value with this:
`GET https://api.connector.mbed.com/v2/endpoints/<my device id>/device/0/name`
it gives me this:
"No notification channel found."
I read the documentation and it says I will need a server to receive a notification call back but I am running node-red at localhost. I just want to receive resource value of my device.
Thanks

teetak01, 2017-02-11:
@masoudr you must set a notification channel, otherwise the server (api.connector.mbed.com) does not know how to respond to asyncronous requests. Use either PUSH or PULL notifications.

FYI: Your first request is syncronous, the server already has that information and can immediately respond with Response 200. When you ask a specific value from a resource, the server has to go and ask the endpoint for that value. This will be handled in Asyncronous mode, first Response will be 202, followed by Response 200 via notification channel.

https://docs.mbed.com/docs/mbed-device-connector-web-interfaces/en/latest/api-reference/#notifications

If you cannot set-up webserver to receive PUSH-notifications (Registered callback channel), then use PULL (Long polling).

masoudr, 2017-02-11:
@teetak01 Thanks for the answer. I get it working with PULL notification. I created 2 flows, the first one sends a GET request with resource path on it for example `GET https://api.connector.mbed.com/v2/endpoints/<my device id>/device/0/name` followed by another with `https://api.connector.mbed.com/notification/pull` with the delay of 1 second. It works but it has some problems sometimes it returns the value of the resource in (async-response format) but sometimes it returns some odd value which does not contain any payload of my resource.How can I optimize it to always respond with resource value? for example, if I make a `GET `request after 3 seconds or more it does not respond the resource value or if I make it less than 1 second it returns 3 times of resource value which only contains one.
I think I am doing Long polling. Actually, I don't have any public server to add to the notification channel, how can I use PUSH notification on my local server and what requests should I use?
Thanks.

yogpan01, 2017-02-16:
Long polling is not recommended and it will give you these issues, if you want a reliable mechanism , you should be using notification callback mechanism.

closed 2017-02-16 by yogpan01

masoudr, 2017-02-16:
You are right, I registered my notification callback channel and it worked better than long polling. But I think it would be better to make long polling fixed because for local servers it is useful. I prefer to set my local web server for using mbed connector rather than a public server, but I think there is no way to use it without the problem.