ARMmbed/mbed-os-example-client #187: Gateway role

finikorg, 2017-02-23:
Can I use frdm-k64f + 802.15.4 shield as a gateway? It has almost the same hardware as a 6lowpan gateway from Firefly, right?

yogpan01, 2017-02-23:
@SeppoTakalo Can you please answer this query ?

SeppoTakalo, 2017-02-23:
Yes.
The example application is currently here: https://github.com/ARMmbed/k64f-border-router

It uses K64F and Atmel RF shield.
In future, the repository will move and more boards will be supported.

finikorg, 2017-02-23:
@SeppoTakalo There is a note there "Note: This Border Router does not support Thread."

SeppoTakalo, 2017-02-23:
For mbed OS 5.4 release, we will also publish Thread mesh networking support.

Current one is just in 6LoWPAN-ND mode.

SeppoTakalo, 2017-02-24:
I'll close this issue as the question is answered.
For follow up questions you can still use this ticket, thanks.

closed 2017-02-24 by SeppoTakalo