ARMmbed/mbed-os-example-client #192: Run mbed client in NUCLEO L073RZ

fwndz, 2017-02-27:
hi,

I would like to run the mbed client example in different evaluation module like L073RZ with ESP8266 wifi module. Which configuration should I change? I already follow the instruction and made some changes in mbed_app.json. However in the area below which parameter should I `change?
```

    "macros": ["MBEDTLS_USER_CONFIG_FILE=\"mbedtls_mbed_client_config.h\""],
    "target_overrides": {
        "*": {
            "target.features_add": ["NANOSTACK", "LOWPAN_ROUTER", "COMMON_PAL"],
            "platform.stdio-baud-rate": 115200,
            "platform.stdio-convert-newlines": true,
            "mbed-mesh-api.6lowpan-nd-channel-page": 0,
            "mbed-mesh-api.6lowpan-nd-channel": 12,
            "mbed-trace.enable": 0
},
```
Where can I enable IPv4? After debugging the example I encountered these error.
```
[Error] easy-connect.h@1,1: unterminated #ifndef
[Error] main.cpp@41,19: a function-definition is not allowed here before '{' token
[Error] main.cpp@112,81: no matching function for call to 'FP1<void, void*>::FP1(easy_connect(bool)::LedResource*, void (easy_connect(bool)::LedResource::*)(void*))'
[Error] main.cpp@112,81: template argument for 'template<class R, class A1> template<class T> FP1<R, A1>::FP1(T*, R (T::*)(A1))' uses local type 'easy_connect(bool)::LedResource'
[Error] main.cpp@112,5:   trying to instantiate 'template<class R, class A1> template<class T> FP1<R, A1>::FP1(T*, R (T::*)(A1))'
[Error] main.cpp@128,9: use of local variable with automatic storage from containing function
[Error] main.cpp@129,9: use of local variable with automatic storage from containing function
[Error] main.cpp@168,66: no matching function for call to 'callback(easy_connect(bool)::LedResource*, void (easy_connect(bool)::LedResource::*)())'
[Error] main.cpp@168,66: template argument for 'template<class T, class U, class R> mbed::Callback<R()> mbed::callback(U*, R (T::*)())' uses local type 'easy_connect(bool)::LedResource'
[Error] main.cpp@168,0: template argument for 'template<class T, class U, class R> mbed::Callback<R()> mbed::callback(U*, R (T::*)())' uses local type 'easy_connect(bool)::LedResource'
[Error] main.cpp@168,15:   trying to instantiate 'template<class T, class U, class R> mbed::Callback<R()> mbed::callback(U*, R (T::*)())'
[Error] main.cpp@178,13: use of local variable with automatic storage from containing function
[Error] main.cpp@178,24: use of local variable with automatic storage from containing function
[Error] main.cpp@185,17: use of local variable with automatic storage from containing function
[Error] main.cpp@186,17: use of local variable with automatic storage from containing function
[Error] main.cpp@228,13: use of local variable with automatic storage from containing function
[Error] main.cpp@263,102: no matching function for call to 'FP1<void, M2MBlockMessage*>::FP1(easy_connect(bool)::BigPayloadResource*, void (easy_connect(bool)::BigPayloadResource::*)(M2MBlockMessage*))'
[Error] main.cpp@263,102: template argument for 'template<class R, class A1> template<class T> FP1<R, A1>::FP1(T*, R (T::*)(A1))' uses local type 'easy_connect(bool)::BigPayloadResource'
[Error] main.cpp@263,5:   trying to instantiate 'template<class R, class A1> template<class T> FP1<R, A1>::FP1(T*, R (T::*)(A1))'
[Error] main.cpp@265,103: no matching function for call to 'FP3<void, const m2m::String&, unsigned char*&, long unsigned int&>::FP3(easy_connect(bool)::BigPayloadResource*, void (easy_connect(bool)::BigPayloadResource::*)(con
st m2m::String&, uint8_t*&, uint32_t&))'
[Error] main.cpp@265,103: template argument for 'template<class R, class A1, class A2, class A3> template<class T> FP3<R, A1, A2, A3>::FP3(T*, R (T::*)(A1, A2, A3))' uses local type 'easy_connect(bool)::BigPayloadResource'
[Error] main.cpp@265,5:   trying to instantiate 'template<class R, class A1, class A2, class A3> template<class T> FP3<R, A1, A2, A3>::FP3(T*, R (T::*)(A1, A2, A3))'
[Error] main.cpp@308,19: a function-definition is not allowed here before '{' token
[Error] main.cpp@313,23: a function-definition is not allowed here before '{' token
[Error] main.cpp@319,12: a function-definition is not allowed here before '{' token
[Warning] main.cpp@304,15: unused variable 'registered' [-Wunused-variable]
```
How can I migrate to the current example which used different board and network stack to what I mentioned above. Is there any simple guideline which I should follow?

Cheers
Fwndz

JanneKiiskila, 2017-02-28:
Hi @fwndz 

please do not change the configuration. In this case it should be enough that you just

mbed deploy
`cp configs/wifi_esp8266_v4.json mbed_app.json`
<favourite editor> `mbed_app.json`
<change the wifi SSID + password to your needs>
`cp configs/eth-wifi-mbedignore .mbedignore`

and then just `mbed compile` as per usual.

The 1st error
`[Error] easy-connect.h@1,1: unterminated #ifndef`

shows you've potentially now modified something from the easy-connect - you should not need to do at at all.

JanneKiiskila, 2017-02-28:
Hi,

@bulislaw @MarceloSalazar @sg- 

seems the support for this board is still a bit unfinished, trying to just compile normal ethernet for that board fails. It doesn't seem to know how to ask for "LWIP". Once you add the "LWIP" to mbed_app.json, it still fails.

```
jankii01@ubuntu:~/mbed/mbed-os-example-client$ git diff mbed_app.json
diff --git i/mbed_app.json w/mbed_app.json
index 11aa0f7..2c8b83e 100644
--- i/mbed_app.json
+++ w/mbed_app.json
@@ -28,7 +28,7 @@
     "macros": ["MBEDTLS_USER_CONFIG_FILE=\"mbedtls_mbed_client_config.h\""],
     "target_overrides": {
         "*": {
-            "target.features_add": ["NANOSTACK", "LOWPAN_ROUTER", "COMMON_PAL"],
+            "target.features_add": ["LWIP", "NANOSTACK", "LOWPAN_ROUTER", "COMMON_PAL"],
             "platform.stdio-baud-rate": 115200,
             "platform.stdio-convert-newlines": true,
             "mbed-mesh-api.6lowpan-nd-channel-page": 0,
```

```
[Error] stm32xx_emac.c@15,1: unknown type name 'ETH_HandleTypeDef'
[Error] stm32xx_emac.c@20,15: unknown type name 'ETH_DMADescTypeDef'
[Error] stm32xx_emac.c@20,47: 'ETH_RXBUFNB' undeclared here (not in a function)
[Error] stm32xx_emac.c@25,15: unknown type name 'ETH_DMADescTypeDef'
[Error] stm32xx_emac.c@25,47: 'ETH_TXBUFNB' undeclared here (not in a function)
[Error] stm32xx_emac.c@30,44: 'ETH_RX_BUF_SIZE' undeclared here (not in a function)
[Error] stm32xx_emac.c@35,44: 'ETH_TX_BUF_SIZE' undeclared here (not in a function)
[Error] stm32xx_emac.c@62,29: unknown type name 'ETH_HandleTypeDef'
[Warning] stm32xx_emac.c@76,5: implicit declaration of function 'HAL_ETH_IRQHandler' [-Wimplicit-function-declaration]
[Error] stm32xx_emac.c@95,14: request for member 'Instance' in something not a structure or union
[Error] stm32xx_emac.c@95,26: 'ETH' undeclared (first use in this function)
[Error] stm32xx_emac.c@96,14: request for member 'Init' in something not a structure or union
[Error] stm32xx_emac.c@96,38: 'ETH_AUTONEGOTIATION_ENABLE' undeclared (first use in this function)
[Error] stm32xx_emac.c@97,14: request for member 'Init' in something not a structure or union
[Error] stm32xx_emac.c@97,28: 'ETH_SPEED_100M' undeclared (first use in this function)
[Error] stm32xx_emac.c@98,14: request for member 'Init' in something not a structure or union
[Error] stm32xx_emac.c@98,33: 'ETH_MODE_FULLDUPLEX' undeclared (first use in this function)
[Error] stm32xx_emac.c@99,14: request for member 'Init' in something not a structure or union
[Error] stm32xx_emac.c@110,14: request for member 'Init' in something not a structure or union
[Error] stm32xx_emac.c@111,14: request for member 'Init' in something not a structure or union
[Error] stm32xx_emac.c@111,29: 'ETH_RXINTERRUPT_MODE' undeclared (first use in this function)
[Error] stm32xx_emac.c@112,14: request for member 'Init' in something not a structure or union
[Error] stm32xx_emac.c@112,35: 'ETH_CHECKSUM_BY_HARDWARE' undeclared (first use in this function)
[Error] stm32xx_emac.c@113,14: request for member 'Init' in something not a structure or union
[Error] stm32xx_emac.c@113,37: 'ETH_MEDIA_INTERFACE_RMII' undeclared (first use in this function)
```
Seems the EMAC / ETH stuff is not finished yet. Same happens also with the ESP8266 as well.

Should this be raised as an issue to mbedOS? This is not a mbed-os-example-client issue, IMHO.

JanneKiiskila, 2017-02-28:
Well,

after some tweaking I got it compiling, but doesn't help as it does not fit into the RAM available on that board.

```
Link: mbed-os-example-client
/usr/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: ./BUILD/NUCLEO_L073RZ/GCC_ARM/mbed-os-example-client.elf section `.text' will not fit in region `FLASH'
/usr/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: ./BUILD/NUCLEO_L073RZ/GCC_ARM/mbed-os-example-client.elf section `.bss' will not fit in region `RAM'
/usr/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: region RAM overflowed with stack
/usr/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 58724 bytes
/usr/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: region `RAM' overflowed by 8788 bytes
collect2: error: ld returned 1 exit status
[ERROR] /usr/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: ./BUILD/NUCLEO_L073RZ/GCC_ARM/mbed-os-example-client.elf section `.text' will not fit in region `FLASH'
/usr/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: ./BUILD/NUCLEO_L073RZ/GCC_ARM/mbed-os-example-client.elf section `.bss' will not fit in region `RAM'
/usr/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: region RAM overflowed with stack
/usr/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 58724 bytes
/usr/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: region `RAM' overflowed by 8788 bytes
collect2: error: ld returned 1 exit status

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /home/jankii01/mbed/mbed-os-example-client/mbed-os/tools/make.py -t GCC_ARM -m NUCLEO_L073RZ --source . --build ./BUILD/NUCLEO_L073RZ/GCC_ARM" in "/home/jankii01/mbed/mbed-os-example-client"
```

`mbed_app.json` was like this

```
jankii01@ubuntu:~/mbed/mbed-os-example-client$ git diff mbed_app.json
diff --git i/mbed_app.json w/mbed_app.json
index 11aa0f7..9b597ea 100644
--- i/mbed_app.json
+++ w/mbed_app.json
@@ -2,7 +2,7 @@
     "config": {
         "network-interface":{
             "help": "Options are ETHERNET, WIFI_ESP8266, WIFI_ODIN, MESH_LOWPAN_ND, MESH_THREAD",
-            "value": "ETHERNET"
+            "value": "WIFI_ESP8266"
         },
         "mesh_radio_type": {
                "help": "options are ATMEL, MCR20",
@@ -28,11 +28,9 @@
     "macros": ["MBEDTLS_USER_CONFIG_FILE=\"mbedtls_mbed_client_config.h\""],
     "target_overrides": {
         "*": {
-            "target.features_add": ["NANOSTACK", "LOWPAN_ROUTER", "COMMON_PAL"],
+            "target.features_add": ["COMMON_PAL"],
             "platform.stdio-baud-rate": 115200,
             "platform.stdio-convert-newlines": true,
-            "mbed-mesh-api.6lowpan-nd-channel-page": 0,
-            "mbed-mesh-api.6lowpan-nd-channel": 12,
             "mbed-trace.enable": 0
         },
         "NUCLEO_F401RE": {
```

I used the `.mbedignore` by copying it from the `configs/eth-wifi_mbedignore`

closed 2017-02-28 by JanneKiiskila

JanneKiiskila, 2017-02-28:
If you want to try the WiFi on that board - this example seemed to compile & fit in,

https://github.com/ARMmbed/mbed-os-example-wifi

fwndz, 2017-02-28:
Hi @JanneKiiskila ,
Thanks for you kind support. I have one quick question, is it possible to send a file by using HTTP POST with this example.
https://github.com/ARMmbed/mbed-os-example-wifi

If yes, can you briefly told me how to implement it with the current libraries provided by mbed.

Cheers

bulislaw, 2017-02-28:
There's no out of box support for that. You'll need to have your own HTTP library and connect it manually to FS.

fwndz, 2017-02-28:
Hi @bulislaw, Thanks for the answer

janjongboom, 2017-03-01:
@fwndz See https://developer.mbed.org/teams/sandbox/code/http-example/, should work on that board too.