ARMmbed/mbed-os-example-client #195: WIFI_ODIN configuration and .mbedignore instructions

bridadan, 2017-02-28:
The instructions to add the `.mbedignore` file for the conflict between LWIP and the mesh RF stacks are at the very bottom of the README. I missed these for the first 10 minutes I was trying to use the WIFI_ODIN configuration.

Perhaps we could break down the steps by the `"network-interface"` setting in the mbed_app.json? And perhaps reference the users to the already available config files in the `/config` folder?

yogpan01, 2017-03-02:
@peknis01 Can you check the documentation bit ?

bridadan, 2017-03-02:
To be clear, most of the information is already in the README. But with a little reorganization I think the experience could be much smoother!

As a user, my thought process is "look for instructions for my configuration -> read those instructions" vs "read most instructions and then search around a bit more if things aren't working". 

Thanks again @yogpan01 @peknis01!

peknis01, 2017-03-03:
Take a look at: https://github.com/ARMmbed/mbed-os-example-client/pull/197

yogpan01, 2017-05-02:
This can be closed as we have now updated the README documentation structure.

closed 2017-05-02 by yogpan01