ARMmbed/mbed-os-example-client #205: Fails to connect with K64F and ESP8266

sarahmarshy, 2017-03-10:
Full log:

Starting mbed Client example in IPv4 mode
[EasyConnect] Using WiFi (ESP8266)
[EasyConnect] Connecting to WiFi iotlab
òò
Starting mbed Client example in IPv4 mode
[EasyConnect] Using WiFi (ESP8266)
[EasyConnect] Connecting to WiFi iotlab
[EasyConnect] Connected to Network successfully
[EasyConnect] IP address 192.168.1.166
[EasyConnect] MAC address 18:fe:34:1c:cf:60

SOCKET_MODE : TCP
Connecting to coap://api.connector.mbed.com:5684
[ERROR:] M2MInterface::NetworkError

[ERROR:] M2MInterface::NetworkError

JanneKiiskila, 2017-03-13:
@sarahmarshy - do you think the linked issue is behind this or not? We have a CI run for this example which runs also over WiFi and it is working (with this exact combination).

I also did a manual test myself as well, using the current top of the master.

```
*   1c7333e Yogesh Pande - Merge pull request #190 from ARMmbed/oob_test_mbed-os-5.4
```

built it with 
```
mbed compile -m K64F -t GCC_ARM -c
```
outcome binary had these sizes:

```
+-----------------------------+--------+-------+-------+
| Module                      |  .text | .data |  .bss |
+-----------------------------+--------+-------+-------+
| Fill                        |    448 |    14 |  2526 |
| Misc                        | 123613 |  2585 |  1660 |
| drivers                     |   1643 |     0 |     0 |
| features/FEATURE_COMMON_PAL |   9595 |    57 |  8361 |
| features/mbedtls            |  73371 |    51 |  8755 |
| features/netsocket          |   4212 |    85 |     0 |
| features/storage            |     42 |     0 |   184 |
| hal                         |    580 |     0 |     8 |
| platform                    |   1652 |    20 |   299 |
| rtos                        |    757 |     4 |     4 |
| rtos/rtx                    |   7221 |    20 |  6871 |
| targets/TARGET_Freescale    |   9878 |    12 |  1056 |
| Subtotals                   | 233012 |  2848 | 29724 |
+-----------------------------+--------+-------+-------+
Allocated Heap: 24580 bytes
Allocated Stack: unknown
Total Static RAM memory (data + bss): 32572 bytes
Total RAM memory (data + bss + heap + stack): 57152 bytes
Total Flash memory (text + data + misc): 236900 bytes
```
and logs show this:

```
Starting mbed Client example in IPv4 mode                                                                                           
[EasyConnect] Using WiFi (ESP8266)                                                                                                  
[EasyConnect] Connecting to WiFi systest                                                                                            
[EasyConnect] Connected to Network successfully                                                                                     
[EasyConnect] IP address 192.168.8.112                                                                                              
[EasyConnect] MAC address 5c:cf:7f:1c:b2:a8                                                                                         
                                                                                                                                    
SOCKET_MODE : TCP                                                                                                                   
Connecting to coap://api.connector.mbed.com:5684                                                                                    
                                                                                                                                    
Registered object successfully!
```

So, it was successfull. I can also see the device via connector.mbed.com.

JanneKiiskila, 2017-03-14:
Hi,

@sarahmarshy  - you could also try to set on the client logs, modify the mbed_app.json - "mbed-trace.enable": 0 to 1, potentially also enabling the TLS logs as well.

soramame21, 2017-03-16:
Hi @sarahmarshy, @JanneKiiskila  This is a duplicated issue. I reported another one, please refer 
https://github.com/ARMmbed/mbed-os-example-client/issues/212

JanneKiiskila, 2017-03-16:
@sarahmarshy  - can you confirm you were also using `mbed compile -m K64F -t ARM -c`?

sarahmarshy, 2017-03-20:
@JanneKiiskila Can you confirm your mbed-os hash? 

Mine is: `88a4baa Merge pull request #3902 from mazimkhan/master` 

Here is the full log with trace enabled:

```
Starting mbed Client example in IPv4 mode
[EasyConnect] Using Ethernet
[EasyConnect] Connection to Network Failed -3004!

Connection to Network Failed - exiting application...

Starting mbed Client example in IPv4 mode
[EasyConnect] Using Ethernet
òòòò
Starting mbed Client example in IPv4 mode
[EasyConnect] Using WiFi (ESP8266)
[EasyConnect] Connecting to WiFi iotlab
[EasyConnect] Connected to Network successfully
[EasyConnect] IP address 192.168.1.192
[EasyConnect] MAC address 5c:cf:7f:1c:cf:60
[DBG ][mClt]: M2MInterfaceFactory::create_object : Name : 3200
[DBG ][mClt]: M2MObject::create_object_instance - id: 0
[DBG ][mClt]: M2MObject::object_instance(inst_id 0)
[DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 5501)
[DBG ][mClt]: M2MResourceInstance::set_value()
[DBG ][mClt]: M2MResourceInstance::is_value_changed() -- true
[DBG ][mClt]: M2MResourceInstance::report()
[DBG ][mClt]: M2MResourceInstance::report() - level 0
[DBG ][mClt]: M2MResourceInstance::report() - mode = 1, is_observable = 1
[DBG ][mClt]: M2MInterfaceFactory::create_object : Name : 3201
[DBG ][mClt]: M2MObject::create_object_instance - id: 0
[DBG ][mClt]: M2MObject::object_instance(inst_id 0)
[DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 5853)
[DBG ][mClt]: M2MResourceInstance::set_value()
[DBG ][mClt]: M2MResourceInstance::is_value_changed() -- true
[DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 5850)
[DBG ][mClt]: M2MInterfaceFactory::create_object : Name : 1000
[DBG ][mClt]: M2MObject::create_object_instance - id: 0
[DBG ][mClt]: M2MObject::object_instance(inst_id 0)
[DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 1)
[DBG ][mClt]: M2MResourceInstance::set_value()
[DBG ][mClt]: M2MResourceInstance::is_value_changed() -- true
[DBG ][mClt]: M2MInterfaceFactory::create_interface - IN
[DBG ][mClt]: M2MInterfaceFactory::create_interface - parameters endpoint name :            92c6504f-75be-437d-8868-8b4e8bc3bdc4
[DBG ][mClt]: M2MInterfaceFactory::create_interface - parameters endpoint type :            test
[DBG ][mClt]: M2MInterfaceFactory::create_interface - parameters life time(in se           cs): 100
[DBG ][mClt]: M2MInterfaceFactory::create_interface - parameters Listen Port : 0
[DBG ][mClt]: M2MInterfaceFactory::create_interface - parameters Binding Mode :            9
[DBG ][mClt]: M2MInterfaceFactory::create_interface - parameters NetworkStack :            1
[DBG ][mClt]: M2MInterfaceFactory::create_interface - Creating M2MInterfaceImpl
[DBG ][evlp]: event_loop_thread
[nefault type: 0
[DBG ][mClt]: M2MObject::oIBG ][mClt]: M2MObject::object_instance(inst_id 0)
[0bject_instance(inst_id 0)
[DBG ][mClt]: M2Mm
Object::object_instance(inst_id 0)
[DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 0)
[DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 1)
[DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 6)
[DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 7)
[DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 8)
[DBG ][mClt]: M2MNsdlInterface::M2MNsdlInterface()
[DBG ][coap]: Coap random msg ID: 32517
[DBG ][mClt]: M2MNsdlInterface::initialize()
[DBG ][mClt]: M2MInterfaceImpl::M2MInterfaceImpl() -IN
[DBG ][mClt]: M2MNsdlInterface::create_endpoint( name 92c6504f-75be-437d-8868-8b           4e8bc3bdc4 type test lifetime 100, domain 86c11bf8-f541-4800-99b7-f9172eafd709,            mode 1)
[DBG ][mClt]: M2MInterfaceImpl::M2MInterfaceImpl() -OUT
[DBG ][mClt]: M2MInterfaceFactory::create_interface - OUT
[... 236 lines cut ...]
[DBG ][mClt]: M2MInterfaceImpl::socket_error: (4), retry (1), reconnecting (1)
[DBG ][mClt]: M2MInterfaceImpl::internal_event : new state 0
[DBG ][mClt]: M2MInterfaceImpl::state_engine
[DBG ][mClt]: M2MInterfaceImpl::state_idle
[DBG ][mClt]: M2MNsdlInterface::stop_timers()
[DBG ][mClt]: stop_listening()
[DBG ][mClt]: M2MInterfaceImpl::socket_error - reconnecting in 30(s), count 2/3
[DBG ][mClt]: M2MInterfaceImpl::timer_expired()
[DBG ][mClt]: M2MInterfaceImpl::internal_event : new state 6
[DBG ][mClt]: M2MInterfaceImpl::state_engine
[DBG ][mClt]: M2MInterfaceImpl::state_register
[DBG ][mClt]: stop_listening()
[DBG ][mClt]: resolve_server_address()
[DBG ][mClt]: send_dns_event()
[DBG ][mClt]: M2MConnectionHandlerPimpl::dns_handler - _socket_state = 0
[DBG ][mClt]: close_socket() - IN
[DBG ][mClt]: close_socket() - status: 0 OUT
[DBG ][mClt]: M2MInterfaceImpl::socket_error: (4), retry (1), reconnecting (1)
[DBG ][mClt]: M2MInterfaceImpl::internal_event : new state 0
[DBG ][mClt]: M2MInterfaceImpl::state_engine
[DBG ][mClt]: M2MInterfaceImpl::state_idle
[DBG ][mClt]: M2MNsdlInterface::stop_timers()
[DBG ][mClt]: stop_listening()
[DBG ][mClt]: M2MInterfaceImpl::socket_error - reconnecting in 45(s), count 3/3
```

JanneKiiskila, 2017-03-29:
My mbed-os.lib was pointing to: 305f5c491e34d86098e9da91b322017549c189ff

JanneKiiskila, 2017-03-29:
Hi,

Can you please specify the version of the compiler used here?

JanneKiiskila, 2017-04-13:
Closing, it should work.

closed 2017-04-13 by JanneKiiskila