ARMmbed/mbed-os-example-client #208: [OoB_5.4.1] [NuMAKER_PFM_NUC472 + AT233 shield + Thread BR] : Pinmap not found.

ashok-rao, 2017-03-14:
Steps to reproduce:
1. import example, switch to OoB branch, update FW on BR with ThreadDynamic.bin from https://github.com/ARMmbed/mbed-os-example-client/tree/oob-test-mbed-os-5.4.1 . Make sure BR is up & running.
2. mbed compile -t ARM -m NUMAKER_PFM_NUC472 --app-config configs\mesh_thread.json
3. Flash target with binary.

Result on serial terminal:

[image]

teetak01, 2017-03-15:
@ashok-rao could you please open an issue to mbed-OS. This not an application issue.

ashok-rao, 2017-03-15:
@SeppoTakalo .. any instructions on how to add support for this target? Maybe something to include in the docs for other users / other platforms too ?

0xc0170, 2017-03-15:
> @ashok-rao could you please open an issue to mbed-OS. This not an application issue.

+1. It would be helpful to provide what pins are used in this application (or better to pinpoint which pins fails). This board might not be compatible with the shield requirements. 
@ashok-rao  Please provide some additional information that would help us to resolve this .

JanneKiiskila, 2017-03-30:
Hi,

@ashok-rao , @0xc0170 , @bulislaw  - any new updates on this case? I moved our JIRA ticket to IOTMORF, as this is not a client issue (unless you give us some instructions how to add something our documentation, but that info anyway needs to come from You first).

Best Regards,

Janne

0xc0170, 2017-03-30:
@ccli8 @cyliangtw Can you have a look at this issue with the pinmap?

cyliangtw, 2017-03-30:
In `mbed_app.json`, it needs to define macros for AtmelRF shield board as:
```
    "macros": 
        ["ATMEL_SPI_MOSI=A2", "ATMEL_SPI_MISO=A1", "ATMEL_SPI_SCLK=A4", "ATMEL_SPI_CS=A3", "ATMEL_SPI_RST=D5", "ATMEL_SPI_SLP=D7", "ATMEL_SPI_IRQ =D9", "ATMEL_I2C_SDA=D14", "ATMEL_I2C_SCL=D15"],    
```

JanneKiiskila, 2017-04-19:
Does the information from @cyliangtw  resolve this issue @ashok-rao ?

teetak01, 2017-04-19:
That does not look like a very permanent solution. That configuration should go to inside mbed OS, not to application mbed_app.json.

cyliangtw, 2017-04-20:
NUC472 board pinout is Arduino R1 type and not got any specific pinout requirement in mbed enable validation criteria. So, NUC472 board SPI pins are different from current Arduino R3 type. 
About Atmel RF driver, it's default ATMEL_SPI_xxx are Arduino R3 type. As a result, we resolve it by mbed_app.json .
Anyway, our future boards will follow R3 type.

closed 2017-10-25 by teetak01