ARMmbed/mbed-os-example-client #209: Commissioning with gateway?

stevew817, 2017-03-14:
Does the mbed gateway support Thread commissioning?

teetak01, 2017-03-15:
@artokin couldyou answer to this?

mikter, 2017-03-15:
Thread stack of the mbed-os-example-client supports all Commissioner commands according to thread specification.

6loWPAN gateway does not support commissioning.
border router in https://github.com/ARMmbed/thread-testapp-private supports commissioning and can be used as commissioner for testing purposes.

@JanneKiiskila  needs to answer any mbed gateway commissioning support in cloud services

JanneKiiskila, 2017-03-15:
Hi,

All cloud client and cloud stuff is not-network-specific and as such doesn't support any comissioning. I do not think it's even in the scope for that.

@stevew817  - please confirm if this answers your question enough or if you need something more?

stevew817, 2017-03-15:
My question was specific to the mbed gateway hardware as made by Firefly. As far as I can see, there are only binaries provided for it, which appear to have no way of configuring the BR to be a Thread commissioner?

mikter, 2017-03-16:
you are right that binary and hardware does not have possibility to act as commissioner.

JanneKiiskila, 2017-03-16:
I believe you need to have the mbedOS partner license / agreement in place in order to access the mesh networking sources - after that you can compile the SW in any way you want. 

I consider this case now to be handled and with your kind permission I would like to close this issue, would you agree?

stevew817, 2017-03-17:
@JanneKiiskila We have a partner agreement, but even with source there is no way for us currently to compile thread-testapp-private for the Firefly gateway. Is that something that will still be coming, or should I give up on using the gateway as a commissioner?

TuomoHautamaki, 2017-03-22:
Currently we don't have plans to provide commissioning support for mbed 6LoWPAN Gateway. Please try k64f with thread-testapp-private.

stevew817, 2017-03-24:
Understood. You can close the ticket.

closed 2017-03-27 by JanneKiiskila