ARMmbed/mbed-os-example-client #211: [OoB_5.4.1] [NUCLEO_F429ZI + AT233 Thread + Thread BR]: Fails to connect/register object.

ashok-rao, 2017-03-15:
Steps to reproduce:
1. Clone repo, switch branch, update example, update FW of BR with ThreadDynamic.bin. Make sure BR is up & running.
2. edit mbed_app.json with Thread config.
3. mbed compile -t ARM -m NUCLEO_F429ZI --app-config configs\mesh_thread.json -c
4. Flash bin to target.

Result:
Fails to connect & register object. Full log (with trace enabled in mesh_thread.json) attached below.
[Thread_log.txt](https://github.com/ARMmbed/mbed-os-example-client/files/844460/Thread_log.txt)

ashok-rao, 2017-03-15:
@SeppoTakalo .. can you please have a look at this? Thanks!

SeppoTakalo, 2017-03-15:
I think you stopped too quickly.
In Thread network, the devices joins mesh before it has public IP address. First connection attempt most probably fails. Second one should already have public IP.

Let it try for at least those 3 times:
`M2MInterfaceImpl::socket_error - reconnecting in 15(s), count 1/3`

I'll forward this issue to Thread team. In mbed OS interface API, we can only get one `.connect()` callback and Thread returns earlier that 6LoWPAN-ND.

CC: @mikter @artokin

ashok-rao, 2017-03-16:
Thanks @SeppoTakalo .. I tried it, but still the results are same. Here is a detailed explanation and full logs attached at appropriate places. Also, I started with a fresh mbed import & I'm using the oob-test-mbed-os-5.4.1 branch.

Here are the SHA's I'm using:
```
C:\Ashok\mbed_CLI\v5_4_OoB\v_5_4_1_OoB\Client_2\mbed-os-example-client>mbed ls
mbed-os-example-client (300045edd1dd)
|- easy-connect (6fb5842becae)
|  |- atmel-rf-driver (57f22763f4d3)
|  |- esp8266-driver (4ed87bf7fe37)
|  |  `- ESP8266\ATParser (269f14532b98)
|  |- mcr20a-rf-driver (d8810e105d7d)
|  `- stm-spirit1-rf-driver (ac7a4f477222)
|- mbed-client (52e65b46dff7)
|  |- mbed-client-c (c739b8cbcc57)
|  |- mbed-client-classic (b9a521dcd0fc)
|  `- mbed-client-mbed-tls (7e1b6d815038)
|- mbed-os (f4864dc6429e)
`- pal (4e46c0ea8706)
```

Here are the steps I took:
```
C:\Ashok\mbed_CLI\v5_4_OoB\v_5_4_1_OoB\Client_2>mbed import https://github.com/ARMmbed/mbed-os-example-client
[mbed] Importing program "mbed-os-example-client" from "https://github.com/ARMmbed/mbed-os-example-client" at latest revision in the current branch
[mbed] Adding library "easy-connect" from "https://github.com/ARMmbed/easy-connect" at rev #6fb5842becae
[mbed] Adding library "easy-connect\atmel-rf-driver" from "https://github.com/ARMmbed/atmel-rf-driver" at rev #57f22763f4d3
[mbed] Adding library "easy-connect\esp8266-driver" from "https://github.com/ARMmbed/esp8266-driver" at rev #4ed87bf7fe37
[mbed] Adding library "easy-connect\esp8266-driver\ESP8266\ATParser" from "https://github.com/ARMmbed/ATParser" at rev #269f14532b98
[mbed] Adding library "easy-connect\mcr20a-rf-driver" from "https://github.com/ARMmbed/mcr20a-rf-driver" at rev #d8810e105d7d
[mbed] Adding library "easy-connect\stm-spirit1-rf-driver" from "https://github.com/ARMmbed/stm-spirit1-rf-driver" at rev #ac7a4f477222
[mbed] Adding library "mbed-client" from "https://github.com/ARMmbed/mbed-client" at rev #52e65b46dff7
[mbed] Adding library "mbed-client\mbed-client-c" from "https://github.com/ARMmbed/mbed-client-c" at rev #c739b8cbcc57
[mbed] Adding library "mbed-client\mbed-client-classic" from "https://github.com/ARMmbed/mbed-client-classic" at rev #b9a521dcd0fc
[mbed] Adding library "mbed-client\mbed-client-mbed-tls" from "https://github.com/ARMmbed/mbed-client-mbed-tls" at rev #7e1b6d815038
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #f4864dc6429e
[mbed] Adding library "pal" from "https://github.com/ARMmbed/pal" at rev #4e46c0ea8706

C:\Ashok\mbed_CLI\v5_4_OoB\v_5_4_1_OoB\Client_2>cd mbed-os-example-client

C:\Ashok\mbed_CLI\v5_4_OoB\v_5_4_1_OoB\Client_2\mbed-os-example-client>mbed update oob-test-mbed-os-5.4.1
[mbed] Updating program "mbed-os-example-client" to branch oob-test-mbed-os-5.4.1
[mbed] Updating library "easy-connect" to rev #6fb5842becae
[mbed] Updating library "easy-connect\atmel-rf-driver" to rev #57f22763f4d3
[mbed] Updating library "easy-connect\esp8266-driver" to rev #4ed87bf7fe37
[mbed] Updating library "easy-connect\esp8266-driver\ESP8266\ATParser" to rev #269f14532b98
[mbed] Updating library "easy-connect\mcr20a-rf-driver" to rev #d8810e105d7d
[mbed] Updating library "easy-connect\stm-spirit1-rf-driver" to rev #ac7a4f477222
[mbed] Updating library "mbed-client" to rev #52e65b46dff7
[mbed] Updating library "mbed-client\mbed-client-c" to rev #c739b8cbcc57
[mbed] Updating library "mbed-client\mbed-client-classic" to rev #b9a521dcd0fc
[mbed] Updating library "mbed-client\mbed-client-mbed-tls" to rev #7e1b6d815038
[mbed] Updating library "mbed-os" to rev #f4864dc6429e
[mbed] Updating library "pal" to rev #4e46c0ea8706

C:\Ashok\mbed_CLI\v5_4_OoB\v_5_4_1_OoB\Client_2\mbed-os-example-client>notepad++ security.h

C:\Ashok\mbed_CLI\v5_4_OoB\v_5_4_1_OoB\Client_2\mbed-os-example-client>notepad++ mbed_app.json

C:\Ashok\mbed_CLI\v5_4_OoB\v_5_4_1_OoB\Client_2\mbed-os-example-client>mbed compile -t ARM -m NUCLEO_F429ZI -c

(snip)

Link: mbed-os-example-client
Elf2Bin: mbed-os-example-client
+-----------------------------+--------+-------+-------+
| Module                      |  .text | .data |  .bss |
+-----------------------------+--------+-------+-------+
| Misc                        | 292495 |   879 |  3044 |
| drivers                     |   3506 |    12 |   184 |
| features/FEATURE_COMMON_PAL |  17138 |   179 |  8360 |
[... 72 lines cut ...]
            "wifi-tx": "PA_11",
            "wifi-rx": "PA_12"
        },
        "UBLOX_EVK_ODIN_W2": {
        "target.device_has_remove": ["EMAC"]
        }
    }
}
```

And here are the output logs. I have left the target to try to connect for >20 mins and if I read the logs right, I think it went through the 3/3 connect cycles three times (9 tries in total).

We are using a RPi as an IPv6 tunnel. From below log, we have a working IPv6 connection. So, green..
[tunnel_ready.txt](https://github.com/ARMmbed/mbed-os-example-client/files/847394/tunnel_ready.txt)

The mbed GW is up and running with a valid IPv6 address as below. So, green..
[mbed_GW.txt](https://github.com/ARMmbed/mbed-os-example-client/files/847395/mbed_GW.txt)

The output from the target: RED.
[NUCLEO_F429ZI_AT233.txt](https://github.com/ARMmbed/mbed-os-example-client/files/847396/NUCLEO_F429ZI_AT233.txt)

(Extra) If github comments becomes too lengthy, here is the log from cmd prmpt:
[cmd_prmpt.txt](https://github.com/ARMmbed/mbed-os-example-client/files/847397/cmd_prmpt.txt)

@mikter @artokin @JanneKiiskila @MarceloSalazar @adbridge @0xc0170 @screamerbg @sg-

JanneKiiskila, 2017-03-16:
Can you also try with another compiler, GCC? We have another issue where some connectivity issues are reported from ARM compiler (ONLY, GCC and IAR working OK).

SeppoTakalo, 2017-03-16:
Also, does it work in `6LoWPAN-ND` mode?

artokin, 2017-03-16:
From mbed-os-5.4 onwards the Thread gateway binary file `gatewayThreadDynamic.bin` is configured to use RF channel 22 and panid 0x0700. There is no need to modify Thread configuration settings. 
Your steps sounds right in "steps to reproduce" chapter, the application should work in Thread mode when building application with configuration from `./configs/mesh_thread.json`.

artokin, 2017-03-16:
Tested mbed-os-example-client (NUCLEO_F429ZI + AT233, GCC_ARM) in our CI. The device was registered successfully in 3 minutes.

ashok-rao, 2017-03-16:
Hello @JanneKiiskila , the result is same as before with GCC. With IAR, I get a memory allocation error at link time.
```
[Warning] pal_plat_rtos.c@680,19: [Pa089]: enumerated type mixed with another enumerated type
[Warning] pal_plat_rtos.c@707,19: [Pa089]: enumerated type mixed with another enumerated type
[Warning] pal_plat_rtos.c@814,19: [Pa089]: enumerated type mixed with another enumerated type
[Warning] pal_plat_rtos.c@905,0: [Pa089]: enumerated type mixed with another enumerated type
Compile [ 99.8%]: test_env.cpp
Compile [100.0%]: pal_plat_network.cpp
Link: mbed-os-example-client
Error[Lp011]: section placement failed
            unable to allocate space for sections/blocks with a total estimated
                      minimum size of 0x3081c bytes (max align 0x8) in
                      <[0x200001b0-0x2002ffff]> (total uncommitted space
                      0x2fe50).
Error[Lp021]: the destination for compressed initializer batch "P2-1" is placed
          at an address that is dependent on the size of the batch, which is
          not allowed when using lz77 compression. Consider using "initialize
          by copy with packing = zeros" (or none) instead.
[ERROR] Error[Lp011]: section placement failed
            unable to allocate space for sections/blocks with a total estimated
                      minimum size of 0x3081c bytes (max align 0x8) in
                      <[0x200001b0-0x2002ffff]> (total uncommitted space
                      0x2fe50).
Error[Lp021]: the destination for compressed initializer batch "P2-1" is placed
          at an address that is dependent on the size of the batch, which is
          not allowed when using lz77 compression. Consider using "initialize
          by copy with packing = zeros" (or none) instead.

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\Ashok\mbed_CLI\v5_4_OoB\v_5_4_1_OoB\Client_2\mbed-os-example-client\mbed-os\tools\make.py -t IAR -m NUCLEO_F429ZI --source . --build .\BUILD\NUCLEO_F429ZI\IAR -c" in "C:\Ashok\mbed_CLI\v5_4_OoB\v_5_4_1_OoB\Client_2\mbed-os-example-client"
---

C:\Ashok\mbed_CLI\v5_4_OoB\v_5_4_1_OoB\Client_2\mbed-os-example-client>
```

@artokin .. I flashed the GW again with ThreadDynamic.bin , reverted mbed_app.json to defaults and used --app-config with mesh_thread.json while building.

**Happy to report that it works now with this.** . This was the exact same setup I was using yesterday too, but it was failing & that is the reason I reported it.
I suggest to remove or make optional the channel settings chapter in the docs to avoid confusions for users overriding these in mbed_app.json AND using the config json too. Thanks.

artokin, 2017-03-16:
Good, we will update the misleading documentation.

karsev, 2017-03-28:
Document updated, closing the issue.

closed 2017-03-28 by karsev