ARMmbed/mbed-os-example-client #212: [OoB_5.4.1] [HEXIWEAR + wifi ESP8266/K64F + wifi ESP8266]: Fails to connect/register object.

soramame21, 2017-03-15:
**HEXIWEAR + wifi ESP8266** or **K64F + wifi ESP8266** cannot connect to mbed Device Connector.

## Executable failed to connect
The same failed terminal prints can be verified for 2 cases.
Case 1 :  **HEXIWEAR + wifi ESP8266** connect to internet through iphone.
Case 2 :  **K64F + wifi ESP8266** connect to internet through iphone.
### Terminal print out (failed)
```
Starting mbed Client example in IPv4 mode
[EasyConnect] Using WiFi (ESP8266)
[EasyConnect] Connecting to WiFi botren01
[EasyConnect] Connected to Network successfully
[EasyConnect] IP address 172.20.10.12
[EasyConnect] MAC address 18:fe:34:08:1b:3f

SOCKET_MODE : TCP
Connecting to coap://api.connector.mbed.com:5684

[ERROR:] M2MInterface::NetworkError

[ERROR:] M2MInterface::NetworkError

[ERROR:] M2MInterface::NetworkError
```
### Build command
```
1.   mbed compile -m HEXIWEAR -t ARM -c
2.   mbed compile -m K64F -t ARM -c
```

## Executable succeeded to connect
When using **_IAR or GCC_ARM toolchain_** to compile, then both cases succeeded the connection. 

### Terminal print out (Succeeded)
```
Starting mbed Client example in IPv4 mode
[EasyConnect] Using WiFi (ESP8266)
[EasyConnect] Connecting to WiFi botren01
[EasyConnect] Connected to Network successfully
[EasyConnect] IP address 172.20.10.12
[EasyConnect] MAC address 18:fe:34:08:1b:3f

SOCKET_MODE : TCP
Connecting to coap://api.connector.mbed.com:5684

Registered object successfully!
```
### Build command
```
1.   mbed compile -m HEXIWEAR -t GCC_ARM -c
2.   mbed compile -m HEXIWEAR -t IAR -c
3.   mbed compile -m K64F -t GCC_ARM -c
4.   mbed compile -m K64F -t IAR -c
```

soramame21, 2017-03-16:
This is a duplicated issue. please look
https://github.com/ARMmbed/mbed-os-example-client/issues/205

JanneKiiskila, 2017-03-29:
Hi @soramame21 

can you please specify the compiler version? I would almost suspect a compiler bug in this case.

toyowata, 2017-04-18:
Hi @JanneKiiskila

I am facing same problem by online compiler (armcc 5.06 build 61).
But, I didn't have any issue by mbed-os-5.3.6 release.

https://github.com/ARMmbed/mbed-os-example-client/releases/tag/mbed-os-5.3.6

soramame21, 2017-04-26:
Hi @JanneKiiskila 

Very sorry for delayed reply. here is my compiler versions.
`arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release)`

`Component: ARM Compiler 5.06 update 4 (build 422)`

yogpan01, 2017-04-26:
This has been verified that it works fine for ARMCC with latest mbedOS 5.4.3 on K64F, please retest and if its still not working, I suggest try with https://github.com/ARMmbed/mbed-os-example-sockets to verify connectivity.

closed 2017-04-26 by yogpan01