ARMmbed/mbed-os-example-client #215: Long-term stability on K64F

stevew817, 2017-03-17:
After a day or so, my mbed test client (K64F + Atmel) disappears from the Device Connector's "connected devices", and needs to be reset to register again. Is this a known issue?

teetak01, 2017-03-20:
Hi  @stevew817  Could you provide some more details on how you have been testing this.

I presume you are using either of the Mesh-networks with latest mbed-os-example-client?

stevew817, 2017-03-23:
@teetak01 I'm using Thread, standard config shipped with mbedOS.

GW: Firefly IOT gateway, binary from this repo
Node: K64F + Atmel, compiled from this repo.

teetak01, 2017-03-23:
@artokin this is something we have to look together. @stevew817 we will investigate this on our side. Thanks for reporting.

artokin, 2017-03-23:
In some cases we have seen similar behaviour but we were not able to reproduce the problem. We will try to reproduce this error in order to get further details.

terhei, 2017-04-12:
Hi @stevew817
We have been trying to reproduce this issue, but so far we haven't seen any unstability with our setup. 
We will continue working with this issue.

teetak01, 2017-04-26:
Still investigating the root cause of this issue.

yogpan01, 2017-08-15:
Hi we have had multiple runs for mbed-client for over 6 days and there has been no disconnections observed. From mbed-client point of view, client is pretty stable over long testing. There has been similar tests done by other developers and have similar findings.
Check the similar finding and update here https://github.com/ARMmbed/mbed-client/issues/392

closed 2017-08-15 by yogpan01