ARMmbed/mbed-os-example-client #227: Project exported error

SYuPing, 2017-04-05:
Hi, 
I wanted to export mbed-os-client-example to uvision ide, but I got some problems...

version information
mbed-os-example-client (ef89c712c070)
|- easy-connect (6fb5842becae)
|  |- atmel-rf-driver (57f22763f4d3)
|  |- esp8266-driver (4ed87bf7fe37)
|  |  `- ESP8266\ATParser (269f14532b98)
|  |- mcr20a-rf-driver (d8810e105d7d)
|  `- stm-spirit1-rf-driver (ac7a4f477222)
|- mbed-client (52e65b46dff7)
|  |- mbed-client-c (c739b8cbcc57)
|  |- mbed-client-classic (b9a521dcd0fc)
|  `- mbed-client-mbed-tls (7e1b6d815038)
|- mbed-os (f4864dc6429e)
`- pal (4e46c0ea8706)

the mbed-cli version what I used is 1.0.0
target K64F

The result of my mbed-cli is

C:\Users\04070\mbed-os-example-client>mbed export -i uvision -m K64F
Scan: .
Traceback (most recent call last):
  File "C:\Users\04070\mbed-os-example-client\mbed-os\tools\project.py", line 24
8, in <module>
    main()
  File "C:\Users\04070\mbed-os-example-client\mbed-os\tools\project.py", line 24
4, in main
    build_profile=profile)
  File "C:\Users\04070\mbed-os-example-client\mbed-os\tools\project.py", line 94
, in export
    build_profile=build_profile, silent=silent)
  File "C:\Users\04070\mbed-os-example-client\mbed-os\tools\project_api.py", lin
e 199, in export_project
    for loc, path in src_paths.iteritems()}
  File "C:\Users\04070\mbed-os-example-client\mbed-os\tools\project_api.py", lin
e 199, in <dictcomp>
    for loc, path in src_paths.iteritems()}
  File "C:\Users\04070\mbed-os-example-client\mbed-os\tools\build_api.py", line
419, in scan_resources
    resources = toolchain.config.load_resources(resources)
  File "C:\Users\04070\mbed-os-example-client\mbed-os\tools\config.py", line 799
, in load_resources
    self.validate_config()
  File "C:\Users\04070\mbed-os-example-client\mbed-os\tools\config.py", line 771
, in validate_config
    raise self.config_errors[0]
tools.config.ConfigException: Attempt to override undefined parameter 'platform.
stdio-baud-rate' in 'application[*]'
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\Users\04070\mbed-os-example-client\mbed-os\t
ools\project.py -i uvision -m K64F --source ." in "C:\Users\04070\mbed-os-exampl
e-client"
---

How to solve this....
Thanks.

teetak01, 2017-04-05:
Hi @SYuPing , looks like an issue in tools/Online compiler. 

@theotherjimmy @sarahmarshy  can you check this?

theotherjimmy, 2017-04-05:
@teetak01 He specifically mentions mbed-cli, so I doubt it relates to the online ecosystem.

theotherjimmy, 2017-04-05:
@SYuPing This is a configuration error. It looks like you are expecting to be able to overwrite the configuration parameter `platform.stdio-baud-rate`, which is not present in your version of mbed-os.

sarahmarshy, 2017-04-05:
There override is in the default mbed_app.json of this example - https://github.com/ARMmbed/mbed-os-example-client/blob/master/mbed_app.json#L32

theotherjimmy, 2017-04-05:
@SYuPing What is your version of mbed-os

SYuPing, 2017-04-06:
@theotherjimmy I just download this project to export, I think the version is 5.4 for this example also my mbed-cli is 1.0.0

theotherjimmy, 2017-04-06:
@SYuPing mbed-os version 5.4 has an mbed_lib.json that contains the parameter that the tools are complaining about not existing. I'm going to try to reproduce this on my machine.

theotherjimmy, 2017-04-06:
I can reproduce this problem. I think that scanning is not working correctly.

theotherjimmy, 2017-04-06:
https://github.com/ARMmbed/mbed-os/pull/3924 fixed this issue

theotherjimmy, 2017-04-06:
Update to mbed-os 5.4.2, and this problem will go away.

theotherjimmy, 2017-04-06:
Pull the branch from this PR: https://github.com/ARMmbed/mbed-os-example-client/pull/222

closed 2017-04-13 by JanneKiiskila