ARMmbed/mbed-os-example-client #232: Border router dropping routes

stevew817, 2017-04-26:
Using the latest GW_Binary image for Thread, I am observing the gateway becoming non-functional after a while, and starting to emit the following when trying to access a public IP from a node:
```[DBG ][ipv6]: XXX ipv6_buffer_route no route to {prefix}::68c7:55d3!```
and when trying to ping the node:
```
[DBG ][icmp]: Build NA
[DBG ][ipv6]: XXX ipv6_buffer_route no route to {prefix}:cae0:ebff:fe15:3829!
[WARN][ipv6]: ipv6_down route fail```

Since it successfully routes the echo request from ethernet to Thread and to the node, but fails to route the echo response from the Thread network over to Ethernet, I think this is possibly some sort of corruption of the mesh -> LAN routing table?

yogpan01, 2017-04-27:
@artokin @mikter Can you please check this issue ?

artokin, 2017-04-28:
Hi @stevew817 ,

Thanks for reporting this error. Would you please provide more details on this issue so that we can try to reproduce the error? 
* What and how many devices were attached to the network?
* How long did it take to get border router to the unresponsive state?
* What kind of traffic was made on network during operation (heavy load?, big packets?)

BR
Arto

stevew817, 2017-04-28:
@artokin 

- 1 device attached to the border router, running mbed master
- Around 10 minutes after resetting the border router and end node
- Sending a ~190 byte CoAP payload every ~4 seconds. If it matters, this is to a NAT64 address, using the standard prefix 64:ff9b::

artokin, 2017-04-28:
Thanks, we'll try to reproduce this issue.

kjbracey-arm, 2017-05-04:
Was destination prefix was distinct from the Thread and Ethernet prefixes?

Presumably the border router received a default route from Ethernet via Router Advertisement at start-up? Has that timed out? I believe there should have been debug prints from both its addition and any timeout.

stevew817, 2017-05-04:
@kjbracey-arm yes, external destination prefix is the NAT64 prefix (64:ff9b::/96). Trying to ping the node, the prefix was the Ethernet prefix, though.

Thanks for the suggestion. Out of precaution, I updated my router firmware, to see whether it could be the router that is messing up its RAs. We'll see whether that helps.

TuomoHautamaki, 2017-05-24:
@stevew817 any update on this, did router firmware update help?

stevew817, 2017-05-24:
I've been traveling, so haven't been able to get back to this yet. Will be back in the office next week.

mikter, 2017-06-07:
@stevew817 can you give update on this issue please?

stevew817, 2017-06-13:
As far as I can tell, a combination of updating the router and moving to a more stable tunnel provider helped. Sorry for raising the alarm!

closed 2017-06-13 by stevew817