ARMmbed/mbed-os-example-client #236: TCP and UDP connection failed

haishengliu, 2017-04-28:
Hello, i'am working on this project.

i'm unlucky to get this message with TCP connection:

[ERR ][tcp ]: DW No Session: dst: 2607:f0d0:2601:52::20, src port: 49157, dst port: 5684

and with UDP connection:

[ERR ][mClt]: M2MConnectionSecurityPimpl::timer_expired - handshake timeout

trace log file for TCP:

=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2017.04.28 15:02:07 =~=~=~=~=~=~=~=~=~=~=~=

[DBG ][icmp]: Build [NA]

[DBG ][mMCp]: PLC MAC Data Req

[DBG ][mMCp]: plc_mac_setup->active_pd_data_request = 0

[DBG ][loND]: Interface ID: 0, ipv6: 2a01:e35:2f47:b380:fec2:3d00:4:cef8

[DBG ][loND]: State Timer CB

[DBG ][loND]: NS Configured

[DBG ][mMCp]: PLC MAC Data Req

[DBG ][mMCp]: plc_mac_setup->active_pd_data_request = 0

[DBG ][loND]: Interface ID: 0, ipv6: 2a01:e35:2f47:b380:fec2:3d00:4:cef8

[INFO][loND]: Address REG OK: 2a01:e35:2f47:b380:fec2:3d00:4:cef8

[DBG ][6Bo ]: ND Ready

[DBG ][6Bo ]: ND BS ready

[DBG ][6Bo ]: ND Ready!!

[DBG ][mMCp]: PLC MAC Data Req

[DBG ][mMCp]: plc_mac_setup->active_pd_data_request = 0

[DBG ][6Bo ]: Accept & Request

[DBG ][MACh]: Register Device

[DBG ][mMIB]: Set 0, mac16 face mac64: fc:c2:3d:00:00:04:ce:f9, 0

[DBG ][6Bo ]: MC DIS Force

[DBG ][mMCp]: PLC MAC Data Req

[DBG ][mMCp]: plc_mac_setup->active_pd_data_request = 0

[INFO][rplc]: DIO from fe80::ff:fe00:face, rank 80

[DBG ][mMCp]: PLC MAC Data Req

[DBG ][mMCp]: plc_mac_setup->active_pd_data_request = 0

[DBG ][icmp]: Build NA

[DBG ][icmp]: Build NA

[DBG ][icmp]: Build NA

[ERR ][mMCp]: PLC MAC tx fail

[... 1767 lines cut ...]
and log file for UDP:

[DBG ][mClt]: M2MConnectionSecurityPimpl::continue_connecting, ret: -26624

[ERR ][mClt]: M2MConnectionSecurityPimpl::timer_expired - handshake timeout

[DBG ][mClt]: handle_connection_error

[DBG ][mClt]: M2MInterfaceImpl::socket_error: (6), retry (1), reconnecting (1)

[DBG ][mClt]: M2MInterfaceImpl::socket_error - start again

[ERROR:] M2MInterface::SecureConnectionFailed

[DBG ][mClt]: M2MInterfaceImpl::internal_event : new state 0

[DBG ][mClt]: M2MInterfaceImpl::state_engine

[DBG ][mClt]: M2MInterfaceImpl::state_idle

[DBG ][mClt]: M2MNsdlInterface::stop_timers()

[DBG ][mClt]: stop_listening()

[DBG ][mClt]: M2MInterfaceImpl::socket_error - reconnecting in 15(s), count 1/3

yogpan01, 2017-04-28:
@SeppoTakalo can you please look into this ? Can we connect mbed Client over TCP in 6LoWPAN?

SeppoTakalo, 2017-05-02:
@haishengliu Can you specify which hardware is used and also which version of mbed OS and client?

yogpan01, 2017-05-04:
Hi @haishengliu , 
Can you tell which location are you testing it from ? Are you able to ping "api.connector.mbed.com" from your test network ? I am suspecting that there might be some network firewall which prevents your device to reach the server.

haishengliu, 2017-05-04:
Hello, 

thank your very much for your response, i'm from France.

ping api.connector.mbed.com gives us 190ms

with the Atmel RF platform, it works. But we’d like to test out customized specific board, we get this error:

As mentioned in the upper post, the ping command to our client board works perfectly, 

I also get this message:

[DBG ][mMCp]: Too Long 134, 110 pa 18 header 0 mic

[ERR ][6lAd]: MCPS Data fail by status 241

Thanks again

Haisheng

yogpan01, 2017-05-04:
There is one "code change" that you can do to see if that helps for UDP case. 
https://github.com/ARMmbed/mbed-client-mbed-tls/blob/master/source/m2mconnectionsecuritypimpl.cpp#L294
In this line, you can change the timeout value from 29000 (29 seconds) to 200000 (200 seconds) just to check if it helps to overcome any network slowness and helps in registering your client in UDP mode.

SeppoTakalo, 2017-05-05:
I would recommend that you update your mbed OS version to latest release.

The TCP support for Nanostack is improved between mbed OS 5.2 and 5.3. TCP server was not supported at all and Nanostack was not buffering any data.

SeppoTakalo, 2017-05-31:
Please try with current mbed OS release.
Re-open if issue is still valid.

closed 2017-05-31 by SeppoTakalo