ARMmbed/mbed-os-example-client #237: building with configs/eth_v4.json results in nanostack/platform/arm_hal_phy.h: No such file or directory

tubaman, 2017-05-01:
### Description
- Type: Bug

Building with

    mbed compile -m K64F -t GCC_ARM -c --app-config configs/eth_v4.json

...fails with the following error: `nanostack/platform/arm_hal_phy.h: No such file or directory`

---------------------------------------------------------------
## Bug

**Version information**
```
mbed-os-example-client (5f27acc88541)
|- easy-connect (6fb5842becae)
|  |- atmel-rf-driver (57f22763f4d3)
|  |- esp8266-driver (4ed87bf7fe37)
|  |  `- ESP8266/ATParser (269f14532b98)
|  |- mcr20a-rf-driver (d8810e105d7d)
|  `- stm-spirit1-rf-driver (ac7a4f477222)
|- mbed-client (f8f0fc8b9321)
|  |- mbed-client-c (c739b8cbcc57)
|  |- mbed-client-classic (f673b8b60779)
|  `- mbed-client-mbed-tls (7e1b6d815038)
|- mbed-os (ed4febefdede)
`- pal (4e46c0ea8706)
```

**Environment details**
I'm on Debian Stretch 64-bit with the following toolchain: gcc-arm-none-eabi-4_9-2015q3

**Expected Behavior**
Building with

    mbed compile -m K64F -t GCC_ARM -c --app-config configs/eth_v4.json

...fails with the following error: `nanostack/platform/arm_hal_phy.h: No such file or directory`

**Steps to Reproduce**

I followed the README

tubaman, 2017-05-01:
Here's a possibly [related issue](https://github.com/ARMmbed/mbed-os-example-client/issues/233)

yogpan01, 2017-05-02:
When using the above config.json , you need to take https://github.com/ARMmbed/mbed-os-example-client/blob/master/configs/eth-wifi-mbedignore into .mbedignore.
It will fix your issue.

loverdeg-ep, 2017-05-02:
Which .mbedignore?

yogpan01, 2017-05-03:
Hi, 
Does this documentation bit help  #239 ?

loverdeg-ep, 2017-05-03:
From #239 I get, rename the file with "ignore" in it's name for the config you are using to ".mbedignore"

Though I'm still confused why the RF drivers aren't removed from the build automatically when using  the ethernet config.

yogpan01, 2017-05-03:
Because  those are cloned from another library known as "easy-connect" which pulls in all the needed RF Driver dependencies without really having the knowledge whether the main application needs it or not.

JanneKiiskila, 2017-05-10:
Closing issue, documentation already improved.
Please create a new ticket, if you think more is needed - but it would most likely entail some tool changes to mbed cli.

closed 2017-05-10 by JanneKiiskila