ARMmbed/mbed-os-example-client #247: [ST - NUCLEO_F429ZI + X-NUCLEO-IDS01A4] : not able to connect to device connector when trace is on

architech-boards, 2017-05-15:
**Description**
- Type: Bug

NUCLEO_F429ZI is able to connect to the https://connector.mbed.com/ with a Spirit module: X-NUCLEO-IDS01A4. But if I change the trace in mbed_apps.json, it does not connect to Device connector anymore.

After the connection with the router (it is another NUCLEO_F429ZI + spirit):
```
[2K[EasyConnect] Connected to Network successfully
SOCKET_MODE : UDP
Connecting to coaps://[2607:f0d0:2601:52::20]:5684
```

When the client tries to connect to the server side I have continuosly this error:

```
[2K[90m[DBG ][mClt]: M2MConnectionSecurityPimpl::start_connecting_non_blocking - handshake ret: 1, ssl state: 2
[2K[90m[DBG ][rplc]: transmit dio, rank: 100
[2K[90m[DBG ][nsif]: socket_callback() sock=1, event=144, interface=1, data len=115
[2K[90m[DBG ][nsif]: SOCKET_TX_DONE, 115 bytes sent
[2K[90m[DBG ][mClt]: receive_handler()
[2K[90m[DBG ][mClt]: receive_handshake_handler()
[2K[90m[DBG ][mClt]: M2MConnectionSecurityPimpl::continue_connecting
[2K[90m[DBG ][mClt]: mbedtls_timing_get_delay
[2K[90m[DBG ][mClt]: mbedtls_timing_get_delay - ret 0
[2K[90m[DBG ][mClt]: receive_from_socket
[2K[90m[DBG ][nsif]: socket_recvfrom(socket=0x20013408) sock_id=1, ret=-3001, addr=[::]:0
[2K[90m[DBG ][mClt]: M2MConnectionSecurityPimpl::continue_connecting, ret: -1000
[2K[90m[DBG ][mClt]: ret -1000
[2K[90m[DBG ][mClt]: We wait for next readable event
```

where I suppose the error codes are:
m2mconnectionhandler.h:        CONNECTION_ERROR_WANTS_READ = -1000
nsapi_types.h:    NSAPI_ERROR_WOULD_BLOCK   = -3001

**Version information**

mbed-os-example-client (34692b03f0f5)
|- easy-connect (6fb5842becae)
|  |- atmel-rf-driver (57f22763f4d3)
|  |- esp8266-driver (4ed87bf7fe37)
|  |  `- ESP8266\ATParser (269f14532b98)
|  |- mcr20a-rf-driver (d8810e105d7d)
|  `- stm-spirit1-rf-driver (ac7a4f477222)
|- mbed-client (f8f0fc8b9321)
|  |- mbed-client-c (c739b8cbcc57)
|  |- mbed-client-classic (f673b8b60779)
|  `- mbed-client-mbed-tls (7e1b6d815038)
|- mbed-os (8d21974ba35e)
`- pal (4e46c0ea8706)

If needed:
nanostack-border-router.git (a9d0ce4294aa)
|- drivers\TARGET_MCUXpresso_MCUS\sal-nanostack-driver-k64f-eth (ee7e8a9a2122)
|- drivers\TARGET_NUCLEO_F429ZI\sal-nanostack-driver-stm32-eth (1369c4a6a454)
|- drivers\atmel-rf-driver (57f22763f4d3)
|- drivers\mcr20a-rf-driver (d8810e105d7d)
|- drivers\sal-stack-nanostack-slip (4a600894da85)
|- drivers\stm-spirit1-rf-driver (bbde15ee85d0)
`- mbed-os (8d21974ba35e)

**Environment details**
Windows, I am using GCC_ARM toolchain.

**Expected Behavior**
Device can be registered even with trace enabled

**Actual Behavior**
device doesn't register

**Steps to Reproduce**

- client:
I applied this changes (suggested by @MarceloSalazar) to the source code: https://github.com/ARMmbed/mbed-client-classic/pull/74/commits/6b536a85a3a998c4a5fad66b4e588097b3952b5e
My mbed_app.json is: https://github.com/architech-boards/mbed-os-example-client/blob/master/mbed_app.json

- router:
I'm using the repo: https://github.com/ARMmbed/nanostack-border-router
trace disabled. My mbed_app.json is: https://github.com/architech-boards/nanostack-border-router/blob/master/mbed_app.json

yogpan01, 2017-05-15:
Hi,
I think its the timing issue in the radio driver implementation which is sensitive to delays. Enabling tracing is causing some delay in the transmission/reception of data packets that is causing this issue.
@kjbracey-arm @SeppoTakalo Have you seen this issue, we have seen similar issues with ESP8266 WiFi drivers , enabling tracing causes the client to stop working.

SeppoTakalo, 2017-05-15:
I see quite a lot of debug prints in the ST Spirit driver, for example https://github.com/ARMmbed/stm-spirit1-rf-driver/blob/master/source/NanostackRfPhySpirit1.cpp#L635

Try disabling the traces from the driver side by defining a flag before including that tracing library
**NanostackRfPhySpirit1.cpp**
```
#define MBED_TRACE_MAX_LEVEL 0
#include "mbed_trace.h"
#define TRACE_GROUP  "SPIRIT"
```

betzw, 2017-05-15:
These calls to trace functions should definitively have been met between conditional compilation.
Pls. find a new version of the file in latest commit.
@architech-boards pls. let us know if the issue persists.

architech-boards, 2017-05-15:
There is an improvement updating NanostackRfPhySpirit1.cpp to the last version and adding the MBED_TRACE_MAX_LEVEL. Without the debug of the spirit the client is connecting to the device connector after some tries. However the connection seems stable even if the issue happens at least once per minute.

architech-boards, 2017-05-16:
I tested it, with APP_TRACE_LEVEL the router works even with the trace enabled. Now the trace is enabled on the client and on the router. The client can connects to the mbed connector after some tries.

architech-boards, 2017-05-16:
Sure! In this way I can't see errors on the console, however I see that lesser output means better connection.

betzw, 2017-05-16:
Well, `info` is actually the second highest tracing level. So you will see less output but you still will see output related to information, warnings and errors.

rspelta, 2017-06-07:
Hi, at the moment on the L476RG we have resolved the problem changing the SPI frequency from 1Mhz to 2Mhz:
>module: SimpleSpirit1.cpp
function: void SimpleSpirit1::init()
_spi.frequency(2000000); // 1MHz // betzw: heuristic value // betzw - NOTE: high frequencies lead to instability of Spirit1

With this modify even the router doesn't show anymore the messages:
>[2K[31m[ERR ][mMCp]: MAC tx fail[0m
[2K[31m[ERR ][6lAd]: MCPS Data fail by status 233[0m

betzw, 2017-06-07:
Thanks @architech-boards (@rspelta) for this information.

Seems as if you are not using the latest version of the driver, though. You might want to upgrade the Spirit1 driver to the latest version on the `master` [branch](https://github.com/ARMmbed/stm-spirit1-rf-driver/tree/master).

Pls. note, that in general this timing kind of issues cannot really be _resolved_, these can merely made less likely to happen, see [here](https://github.com/ARMmbed/mbed-os-example-client/pull/248#issuecomment-305477912).

betzw, 2017-09-05:
Can this be closed?

architech-boards, 2017-09-07:
Yes, thanks

closed 2017-09-07 by MarceloSalazar