ARMmbed/mbed-os-example-client #257: mbed assertation failed: 0 == ret in NanostackInterface.cpp

haishengliu, 2017-06-07:
Hello,
It seems working when i followed this tutorial. However, I get this message after running a while : 

mbed assertation failed: 0 == ret, file: ../mbed-os/features/nanostack/FEATURE_NANOSTACK/nanostack-interface/NanostackInterface.cpp, line 213

I posted this issue on the forum also:

https://developer.mbed.org/questions/78214/mbed-assertation-failed-0-ret-in-Nanosta/

Completely no idea about the message. Could you guys take a look at this please?

Thank you in advance

Haisheng

Ps: here is the debug message:

Starting mbed Client example...

Using Mesh

Connecting to Mesh..

Connected to Network successfully

IP address 2a01:e35:2f47:b380:xxxx:xxxx:xxxx:

SOCKET_MODE : UDP

Connecting to coaps:[2607:f0d0:2601:52::20]:5684

Registered object successfully!

led_execute_callback pattern=500:500:500:500:500:500:500

Resource: 3201/0/5850 executed

Payload:

led_execute_callback pattern=500:500:500:500:500:500:500

Resource: 3201/0/5850 executed

Payload:

led_execute_callback pattern=500:500:500:500:500:500:500

Resource: 3201/0/5850 executed

Payload:

led_execute_callback pattern=500:500:500:500:500:500:500

Resource: 3201/0/5850 executed

Payload:

led_execute_callback pattern=500:500:500:500:500:500:500

Resource: 3201/0/5850 executed

Payload:

mbed assertation failed: 0 == ret, file: ../mbed-os/features/nanostack/FEATURE_NANOSTACK/nanostack-interface/NanostackInterface.cpp, line 21

yogpan01, 2017-06-20:
@TuomoHautamaki @SeppoTakalo Can you please look into this issue ?

artokin, 2017-06-20:
Hi,

There is error when closing a socket and I believe this error is corrected in the latest version of mbed-os that was released today.

Would you please try with the latest version of the application? Please remember to use "mbed deploy" after pull.

Best Regards,
Arto

TuomoHautamaki, 2017-07-04:
Hi @haishengliu , have you tried with the latest mbed-os version?

Regards,
Tuomo

haishengliu, 2017-07-04:
Hi, Tuomo

Thanks for your reply, it seems working. However, not often,  I get this error message on the border router side:
[INFO][6Bo ]: Received MLE Advertisement from fe80::fec2:3d00:4:cefa
[INFO][6Bo ]: Received MLE Advertisement from fe80::fec2:3d00:4:cefa
[ERR ][buff]: buffer_get failed: alloc(zd)
[ERR ][pIP6]: RX: out of memory
[ERR ][buff]: buffer_get failed: alloc(zd)
[ERR ][pIP6]: RX: out of memory
[ERR ][buff]: buffer_get failed: alloc(zd)
[ERR ][pIP6]: RX: out of memory

The border router is described on : 
https://github.com/ARMmbed/nanostack-border-router

Thanks 
haisheng

TuomoHautamaki, 2017-07-05:
Hi Haisheng,

As the original issue is now solved, this one should be now closed.
What comes to the border router issue, we have some improvement/fix coming in that area but would be good to get your detailed test steps and logs so that we can then verify issue being solved. So, please raise a new issue under border router repository. Thank you!

Regards,
Tuomo

closed 2017-07-05 by artokin