ARMmbed/mbed-os-example-client #261: Can't connect using Ethernet in IPv6 mode on K64F (oob testing for mbed OS 5.5 RC2)

maclobdell, 2017-06-14:
### Description
- Type: Bug

I know that this is not necessarily a client issue, but I want to enter it for reference.  Can move it over to another location later.  

I verified that I have IPv6 service using http://test-ipv6.com/.  In the past I have had this demo application working.  Something seems to have changed.  

I tested both the mbed-os-5.5.0-rc2-oob and mbedos55-oob branches.  I get the same result.

```
Starting mbed Client example
[EasyConnect] Using Ethernet
[EasyConnect] Connection to Network Failed -3010!

Connection to Network Failed - exiting application...
```

Here is the mbed_app.json file used.

```
{
    "config": {
        "network-interface":{
            "help": "Options are ETHERNET, WIFI_ESP8266, WIFI_ODIN, MESH_LOWPAN_ND, MESH_THREAD",
            "value": "ETHERNET"
        }
    },
    "macros": ["MBEDTLS_USER_CONFIG_FILE=\"mbedtls_mbed_client_config.h\""],
    "target_overrides": {
        "*": {
            "target.features_add": ["LWIP", "COMMON_PAL"],
            "platform.stdio-baud-rate": 115200,
            "platform.stdio-convert-newlines": true,
            "lwip.ipv4-enabled": false,
            "lwip.ipv6-enabled": true,
            "mbed-trace.enable": 0
        },
            "UBLOX_EVK_ODIN_W2": {
            "target.device_has_remove": ["EMAC"]
        }
    }
}
```

teetak01, 2017-06-14:
As you said this clearly is not Client-issue. Such defects should go straight to mbed OS.

https://github.com/ARMmbed/mbed-os/issues/4433

closed 2017-06-14 by teetak01

teetak01, 2017-06-15:
@maclobdell also something what I have previously discovered (not related to this), is that even when the IPv6 test in browser "passes", it still does not guarantee that the network routes the IPv6 traffic correctly. 
In this case already the Connect() call is giving a fail. Could you report the issue to mbed OS, with also details on the Board and Cross-compiler version used. It seems that above fix should have been already in mbed OS so there might still be something else.