ARMmbed/mbed-os-example-client #274: Add an example of how to collect time series data from connector.mbed.com using Telegraf or Prometheus

amq, 2017-07-08:
### Description
- Type: Question

---------------------------------------------------------------

## Question

mbed Device Connector doesn't provide persistence, so an external metric-collecting tool is needed. You already have examples for AWS IoT, Azure IoTHub and others. I'd suggest creating one more example which would include an open-source self-hosted solution like Telegraf or Prometheus.

yogpan01, 2017-07-11:
Hi @amq , is this a requirement that you would like to see on the Server side , if yes, then @szysas can you please check this request ?

amq, 2017-07-13:
@yogpan01 in other words, I'm looking for a way to collect and store the time series data from the mbed connector using an open-source tool.

amq, 2017-07-23:
ie, extend https://github.com/ARMmbed/time_series_database_doc with an open-source solution in addition to Amazon and Azure.

JanneKiiskila, 2017-08-16:
@MarceloSalazar  @senthilr - something you could be looking at?

teetak01, 2017-10-25:
This issue really has nothing to do with Client-side functionality. I will close this.

closed 2017-10-25 by teetak01