ARMmbed/mbed-os-example-client #284: Meet mutex not allowed in ISR context error in debug build

ccli8, 2017-07-25:
### Description
- Type: Bug

---------------------------------------------------------------
## Bug

**Target**
K64F

**Version information**
mbed-os-example-client (mbed-os-5.5.2)
```
mbed-os-example-client (868f1585f40e)
|- easy-connect (cc471beb7e26)
|  |- atmel-rf-driver (0ae76ce17ae5)
|  |- esp8266-driver (918caa01b4cd)
|  |  `- ESP8266\ATParser (269f14532b98)
|  |- mcr20a-rf-driver (d5905fefa54c)
|  `- stm-spirit1-rf-driver (09e4b9664de1)
|- mbed-client (31e5ce203cc0)
|  |- mbed-client-c (ecfa619e42b2)
|  |- mbed-client-classic (4e66929607c3)
|  `- mbed-client-mbed-tls (7e1b6d815038)
|- mbed-os (8828635da469)
`- pal (60ce64d5ec35)
```

**Steps to Reproduce**
Building with default `develop` profile,
```
mbed compile -m K64F -t ARM
```
I get correct result:
```
Starting mbed Client example
[EasyConnect] IPv4 mode
[EasyConnect] Using WiFi (ESP8266) 
[EasyConnect] Connecting to WiFi VVVVVVVVVV
[EasyConnect] Connected to Network successfully
[EasyConnect] MAC address a0:20:a6:09:95:db
[EasyConnect] IP address 192.168.43.196

SOCKET_MODE : TCP
Connecting to coap://api.connector.mbed.com:5684

Registered object successfully!
```
But then building with `debug` profile,
```
mbed compile -m K64F -t ARM --profile debug
```
I meet mutex not allowed in ISR context error:
<pre>
Starting mbed Client example
[EasyConnect] IPv4 mode
[EasyConnect] Using WiFi (ESP8266) 
[EasyConnect] Connecting to WiFi VVVVVVVVVV
[EasyConnect] Connected to Network successfully
[EasyConnect] MAC address a0:20:a6:09:95:db
[EasyConnect] IP address 192.168.43.196

SOCKET_MODE : TCP
Connecting to coap://api.connector.mbed.com:5684
<b>
Mutex 20001300 error -6: Not allowed in ISR context
</b>
</pre>

yogpan01, 2017-07-25:
Hi, Thanks for reporting.
We will look into this issue.

anttiylitokola, 2017-08-25:
Hi,

Tried with the latest codes but I couldn't see such problem anymore. Could you try again with the latest code? This might be related to this issue:
https://github.com/ARMmbed/mbed-os/issues/4904

ccli8, 2017-08-28:
@anttiylitokola I tried with mbed OS https://github.com/ARMmbed/mbed-os/#3e00a22d78f88e989d12d39498a8c621fddc490b, but I still meet this problem.

anttiylitokola, 2017-09-07:
I was able to reproduce this issue now and comments are available here:
https://github.com/ARMmbed/mbed-os-example-client/issues/303
I'll close this issue now so let's follow the progress in #303.

closed 2017-09-07 by anttiylitokola