ARMmbed/mbed-os-example-client #303: Runtime error with WiFi & mbed-os-5.5

betzw, 2017-09-05:
### Description
- Type: Bug

---------------------------------------------------------------
## Bug

**Version information**
mbed-os-example-client (6d49e5370f8a)
|-  mbed-os (4c256f045961)
...

**Environment details**
- GCC_ARM v6.3.1

**Actual Behavior**
```
Starting mbed Client example
[EasyConnect] IPv4 mode
[EasyConnect] Using WiFi (X-NUCLEO-IDW01M1)
[EasyConnect] Connecting to WiFi crespan
[EasyConnect] Connected to Network successfully
[EasyConnect] MAC address 00:80:E1:B8:3F:4A
[EasyConnect] IP address 192.168.1.2

SOCKET_MODE : TCP
Connecting to coap://api.connector.mbed.com:5684
Mutex 0x20001bfc error -6: Not allowed in ISR context
```

MarceloSalazar, 2017-09-05:
@betzw thanks for raising this. Could you share the link to repo/branch of modified easy-connect, so are able to reproduce the app in our side?

betzw, 2017-09-06:
- [easy-connect](https://github.com/betzw/easy-connect/tree/betzw_wb_github)
- [mbed-os-example-client](https://github.com/betzw/mbed-os-example-client/tree/betzw_wifi_wb)

anttiylitokola, 2017-09-06:
@betzw, Does this happen only if you build with the debug profile like described in this ticket https://github.com/ARMmbed/mbed-os-example-client/issues/284 ?

betzw, 2017-09-06:
@anttiylitokola, correct! But this might simply due to having the ISR check disabled in non-`debug` profiles!?!

anttiylitokola, 2017-09-06:
Yes, MBED_TRAP_ERRORS_ENABLED macro is enabled only in debug builds.

betzw, 2017-09-06:
Anyway, it's a bug, isn't it?

betzw, 2017-09-06:
While looking at the call stack in the debugger it seems that this bug might also be

anttiylitokola, 2017-09-07:
Yep, it's a bug. Back trace reveals the problem. [trace.txt](https://github.com/ARMmbed/mbed-os-example-client/files/1284409/trace.txt). 
Problem here is that "platform_enter_critical" used by the event loop is implemented as a mutex, which means you can only make event loop calls from a thread. 
mbed-os 5.6 release will contain "share event queues" which should fix this problem.
https://github.com/ARMmbed/mbed-os/pull/4406

betzw, 2017-09-07:
I have seen that https://github.com/ARMmbed/mbed-os/pull/4406 has already been merged to the `master` branch. Do you think a test with the mbed-os `master` branch would make sense already now?

betzw, 2017-09-07:
OK, I have tried this out and unfortunately my problem is **not** yet solved: 
`event_core_get()` still calls `platform_enter_critical()` which in turn tries to acquire a mutex!

anttiylitokola, 2017-09-07:
m2mconnectionhandler class must be first modified to use this new shared event queue feature. I'll try this locally and propose a PR so you can easily see what needs to be changed.

MarceloSalazar, 2017-09-12:
@anttiylitokola have you had chance to submit a PR?
If possible, we'd like to include this as part of OOB for 5.6.

anttiylitokola, 2017-09-13:
@MarceloSalazar , Unfortunately this change involves also other components to be updated and that's why the change is not ready yet,

screamerbg, 2017-10-05:
@anttiylitokola any update on this?

screamerbg, 2017-10-12:
Bump @anttiylitokola

anttiylitokola, 2017-10-12:
@screamerbg, Unfortunately no updates on this one. There has been lot of other tasks ongoing so I haven't been able to check this further.

JanneKiiskila, 2017-10-23:
True ARM Internal REF is:  IOTCLT-1882

JanneKiiskila, 2018-01-16:
This has been fixed now, but we need to make a new release (which will contain many other things as well). So, it's pending the bigger 1.3 release now, we're trying to get it done as soon as possible.

betzw, 2018-01-16:
Is it already on the master branch?

teetak01, 2018-01-16:
@JanneKiiskila this requires backport fix to public PAL.

teetak01, 2018-05-03:
@betzw this has been fixed in the public https://github.com/armmbed/mbed-cloud-client. The components what this Legacy application uses are deprecated and unlikely to be updated anymore.

See https://github.com/armmbed/mbed-cloud-client-example

closed 2018-05-21 by betzw