ARMmbed/mbed-os-example-client #306: Generally Instability on REALTEK_RTL8195AM

bridadan, 2017-09-13:
### Description
- Type: Bug
- Related Issue: `#abc`

---------------------------------------------------------------
## Bug

This bug smells like a stack overflow issue, hard to say for sure without diving deeper.

**Version information**
mbed OS 5.6 RC1

**Environment details**
Using GCC 6.3.1

**Expected Behavior**
Should be able to make requests to the device from mbed Connector.

**Actual Behavior**
If the program is built with the `release` profile, it connects but I can't issue a GET request to resource /3200/0/5501. The device appears to crash when the request is made. The serial output from the device stops and it disappears from the mbed Connector API portal.

If the program is built with the `debug` profile, it connects but I have trouble with making PUT requests to resource /3201/0/5853. The device appears to crash in a similar way as previously mentioned.

**Steps to Reproduce**
You will also need a pending fix to easy-connect: https://github.com/ARMmbed/easy-connect/pull/44

```
mbed import https://github.com/ARMmbed/mbed-os-example-client
cd mbed-os-example-client/mbed-os
mbed update mbed-os-5.6.0-rc1
cd ../
mbed compile -m REALTEK_RTL8195AM -t GCC_ARm
```

Attempt to PUT to /3201/0/5853 and the device should crash

bridadan, 2017-09-13:
Seems to be ok when using the IAR toolchain. There is a known issue with the ARM 5 toolchain however and it doesn't build.

So not sure if this is something that can be fixed in Client, might need to be passed to Realtek?

yogpan01, 2017-09-14:
@bridadan This should be raised to mbed-os as board support is coming from OS side, the client is agnostic to underlying HW. Can you please raise issue directly to mbed-os and close it here ? We have opened several issue related with Realtek on mbed-os but seems that no one is really looking into this. Here's one example https://github.com/ARMmbed/mbed-os/issues/4683 , I think mbed-os team needs to involve Realtek with this problem otherwise we cannot fix the issue on client end.

bridadan, 2017-09-14:
@yogpan01 Agreed, thanks for confirming. Really just wanted to double check with the Client team since that was the "entry point" in this case. I'll double check with a few other folks and then move the issue. Thanks!

bridadan, 2017-09-14:
After talking it over with the OOB folks, we're gonna leave this one up for a bit as an integration status for the platform. When some of the existing issues logged against mbed OS get resolved we can check again.

MarceloSalazar, 2017-09-14:
@samchuarm @Archcady please be aware of this issue found during OOB.

samchuarm, 2017-09-17:
#5044

samchuarm, 2017-09-17:
https://github.com/ARMmbed/mbed-os/pull/5044

bridadan, 2017-09-19:
Ok great, hopefully this helps with stability!

@MarceloSalazar I'm away from the office this week so I can't test this at the moment. I can check it when I get back if you like.

MarceloSalazar, 2017-09-23:
@dlfryar @yennster could either of you please help to verify this with rc2?

JanneKiiskila, 2017-12-20:
Closing this one out now, we have some fixes coming and other, more specific errors on Mbed OS -side.

closed 2017-12-20 by JanneKiiskila