ARMmbed/mbed-os-example-client #309: Compilation errors with K64 + Ethernet

vijhar01, 2017-09-19:
This is the OOB testing results for k64 + Ethernet compiled with gcc-arm. 
Test passes without and failures
However, reporting the following compilation warnings received as below:

PS C:\mbed-os-example-client\configs> mbed compile -m K64F -t GCC_ARM -c
Building project mbed-os-example-client (K64F, GCC_ARM)
Scan: ..
Scan: env
Scan: mbed
Scan: FEATURE_LWIP
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_STORAGE
Compile [  0.2%]: BufferedPrint.c

-----
------
Compile [ 98.6%]: rtc_api.c
Compile [ 98.8%]: pal_network.c
[Warning] pal_errors.h@42,41: left shift of negative value [-Wshift-negative-value]
[Warning] pal_errors.h@53,65: left shift of negative value [-Wshift-negative-value]
[Warning] pal_errors.h@56,65: left shift of negative value [-Wshift-negative-value]
[Warning] pal_errors.h@69,65: left shift of negative value [-Wshift-negative-value]
[Warning] pal_errors.h@91,65: left shift of negative value [-Wshift-negative-value]
Compile [ 99.0%]: pal_rtos.c
[Warning] pal_errors.h@42,41: left shift of negative value [-Wshift-negative-value]
[Warning] pal_errors.h@53,65: left shift of negative value [-Wshift-negative-value]
[Warning] pal_errors.h@56,65: left shift of negative value [-Wshift-negative-value]
[Warning] pal_errors.h@69,65: left shift of negative value [-Wshift-negative-value]
[Warning] pal_errors.h@91,65: left shift of negative value [-Wshift-negative-value]
Compile [ 99.3%]: pal_init.c
[Warning] pal_errors.h@42,41: left shift of negative value [-Wshift-negative-value]
[Warning] pal_errors.h@53,65: left shift of negative value [-Wshift-negative-value]
[Warning] pal_errors.h@56,65: left shift of negative value [-Wshift-negative-value]
[Warning] pal_errors.h@69,65: left shift of negative value [-Wshift-negative-value]
[Warning] pal_errors.h@91,65: left shift of negative value [-Wshift-negative-value]
Compile [ 99.5%]: pal_plat_network.cpp
[Warning] pal_plat_network.cpp@589,70: 'void Socket::attach(mbed::Callback<void()>)' is deprecated: The behaviour of Soc
ket::attach differs from other attach functions in mbed OS and has been known to cause confusion. Replaced by Socket::si
gio. [since mbed-os-5.4] [-Wdeprecated-declarations]
[Warning] pal_plat_network.cpp@614,45: 'void Socket::attach(mbed::Callback<void()>)' is deprecated: The behaviour of Soc
ket::attach differs from other attach functions in mbed OS and has been known to cause confusion. Replaced by Socket::si
gio. [since mbed-os-5.4] [-Wdeprecated-declarations]
[Warning] pal_plat_network.cpp@729,35: 'void Socket::attach(mbed::Callback<void()>)' is deprecated: The behaviour of Soc
ket::attach differs from other attach functions in mbed OS and has been known to cause confusion. Replaced by Socket::si
gio. [since mbed-os-5.4] [-Wdeprecated-declarations]
Compile [ 99.8%]: pal_plat_rtos.cpp
[Warning] pal_plat_rtos.cpp@251,13: unused variable 'platStatus' [-Wunused-variable]
[Warning] pal_plat_rtos.cpp@252,12: unused variable 'actualOutputLen' [-Wunused-variable]
[Warning] pal_plat_rtos.cpp@422,53: NULL used in arithmetic [-Wpointer-arith]
Compile [100.0%]: test_env.cpp
Link: mbed-os-example-client
Elf2Bin: mbed-os-example-client
+----------------------------------+--------+-------+-------+
| Module                           |  .text | .data |  .bss |
+----------------------------------+--------+-------+-------+
| [fill]                           |    526 |    13 |  2181 |
| [lib]/libc.a                     |  47709 |  2472 |    89 |
| [lib]/libgcc.a                   |   7488 |     0 |     0 |
| [lib]/libnosys.a                 |     32 |     0 |     0 |
| [lib]/libstdc++.a                |   7993 |    20 |   204 |
| [lib]/misc                       |    248 |     8 |    28 |
| easy-connect/esp8266-driver      |    126 |     0 |     0 |
| main.o                           |   7123 |    18 |   322 |
| mbed-client/mbed-client-c        |   5397 |    32 |     0 |
| mbed-client/mbed-client-classic  |   3408 |     3 |    92 |
| mbed-client/mbed-client-mbed-tls |   1534 |     0 |    21 |
| mbed-client/source               |  31229 |     0 |     8 |
| mbed-os/drivers                  |   1099 |     0 |     0 |
| mbed-os/features                 | 130795 |   262 | 63918 |
| mbed-os/hal                      |    748 |     0 |    24 |
| mbed-os/platform                 |   1655 |    20 |   324 |
| mbed-os/rtos                     |  13055 |   180 |  5728 |
| mbed-os/targets                  |  12137 |    12 |  1028 |
| pal/Source                       |   2934 |    32 |   993 |
| Subtotals                        | 275236 |  3072 | 74960 |
+----------------------------------+--------+-------+-------+
Total Static RAM memory (data + bss): 78032 bytes
Total Flash memory (text + data): 278308 bytes

Image: ..\BUILD\K64F\GCC_ARM\mbed-os-example-client.bin

ashok-rao, 2017-09-19:
@teetak01 .. compiler warnings may not be critical to fix right now.. but tagging you just FYI & for tracking.

MarceloSalazar, 2017-09-23:
Opening to keep track of this task and remove warnings when there is a chance.

MarceloSalazar, 2017-09-23:
Agreed to close, and this kind of improvements would be delivered in a separate/internal repo.

closed 2017-09-23 by MarceloSalazar