ARMmbed/mbed-os-example-client #310: Test failure of example with k64 + Ethernet arm compiler

vijhar01, 2017-09-19:
This is the OOB testing results for k64 + Ethernet compiled with arm compiler
Test failed. Here are details: 
-------------------------------

**PS C:\> mbed import https://github.com/ARMmbed/mbed-os-example-client**
[mbed] Importing program "bed-os-example-client" from "https://github.com/ARMmbed/mbed-os-example-client" at latest revision in the current b
ranch
[mbed] Adding library "easy-connect" from "https://github.com/ARMmbed/easy-connect" at rev #bf821b0695cc
[mbed] Adding library "easy-connect\atmel-rf-driver" from "https://github.com/ARMmbed/atmel-rf-driver" at rev #3cc6010188c0
[mbed] Adding library "easy-connect\esp8266-driver" from "https://github.com/ARMmbed/esp8266-driver" at rev #fdb44a4bedbb
[mbed] Adding library "easy-connect\esp8266-driver\ESP8266\ATParser" from "https://github.com/ARMmbed/ATParser" at rev #a2cb09990f11
[mbed] Adding library "easy-connect\mcr20a-rf-driver" from "https://github.com/ARMmbed/mcr20a-rf-driver" at rev #9aac10474702
[mbed] Adding library "easy-connect\stm-spirit1-rf-driver" from "https://github.com/ARMmbed/stm-spirit1-rf-driver" at rev #bee61bbe6807
[mbed] Adding library "mbed-client" from "https://github.com/ARMmbed/mbed-client" at rev #31e5ce203cc0
[mbed] Adding library "mbed-client\mbed-client-c" from "https://github.com/ARMmbed/mbed-client-c" at rev #ecfa619e42b2
[mbed] Adding library "mbed-client\mbed-client-classic" from "https://github.com/ARMmbed/mbed-client-classic" at rev #4e66929607c3
[mbed] Adding library "mbed-client\mbed-client-mbed-tls" from "https://github.com/ARMmbed/mbed-client-mbed-tls" at rev #7e1b6d815038
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #98ba8acb83cf
[mbed] Adding library "pal" from "https://github.com/ARMmbed/pal" at rev #60ce64d5ec35
PS C:\>
PS C:\> cd mbed-os-example-client
**PS C:\mbed-os-example-client> mbed update mbed-os-5.6-oob**
[mbed] Updating program "mbed-os-example-client" to branch mbed-os-5.6-oob
[mbed] Updating library "easy-connect" to rev #d6b91d22a6e9
[mbed] Updating library "easy-connect\atmel-rf-driver" to rev #3cc6010188c0
[mbed] Updating library "easy-connect\esp8266-driver" to rev #fdb44a4bedbb
[mbed] Updating library "easy-connect\esp8266-driver\ESP8266\ATParser" to rev #a2cb09990f11
[mbed] Updating library "easy-connect\mcr20a-rf-driver" to rev #9aac10474702
[mbed] Updating library "easy-connect\stm-spirit1-rf-driver" to rev #bee61bbe6807
[mbed] Updating library "mbed-client" to rev #31e5ce203cc0
[mbed] Updating library "mbed-client\mbed-client-c" to rev #ecfa619e42b2
[mbed] Updating library "mbed-client\mbed-client-classic" to rev #4e66929607c3
[mbed] Updating library "mbed-client\mbed-client-mbed-tls" to rev #7e1b6d815038
[mbed] Updating library "mbed-os" to rev #a4056fb8bf3e
[mbed] Updating library "pal" to rev #60ce64d5ec35
PS C:\mbed-os-example-client> pwd

Path
----
C:\mbed-os-example-client

PS C:\mbed-os-example-client> ls

    Directory: C:\mbed-os-example-client

Mode                LastWriteTime         Length Name
----                -------------         ------ ----
d-----        9/18/2017  10:44 PM                .github
d-----        9/18/2017  10:44 PM                configs
d-----        9/18/2017  10:44 PM                docs
d-----        9/18/2017  10:48 PM                easy-connect
d-----        9/18/2017  10:44 PM                GW_Binary
d-----        9/18/2017  10:44 PM                mbed-client
d-----        9/18/2017  10:48 PM                mbed-os
d-----        9/18/2017  10:47 PM                pal
-a----        9/18/2017  10:44 PM            319 .gitignore
-a----        9/18/2017  10:48 PM              8 .mbed
-a----        9/18/2017  10:44 PM           9215 apache-2.0.txt
-a----        9/18/2017  10:44 PM           2547 build_all.sh
-a----        9/18/2017  10:44 PM         124269 clicks.png
-a----        9/18/2017  10:48 PM             83 easy-connect.lib
-a----        9/18/2017  10:44 PM           4369 Jenkinsfile
-a----        9/18/2017  10:44 PM            135 LICENSE
-a----        9/18/2017  10:44 PM          15414 main.cpp
-a----        9/18/2017  10:44 PM             82 mbed-client.lib
-a----        9/18/2017  10:48 PM             78 mbed-os.lib
-a----        9/18/2017  10:44 PM           3464 mbedtls_mbed_client_config.h
-a----        9/18/2017  10:44 PM           1590 mbed_app.json
-a----        9/18/2017  10:44 PM           2291 mbed_client_config.h
-a----        9/18/2017  10:47 PM           1374 mbed_settings.py
[... 495 lines cut ...]
[Warning] core_cm4.h@28,0:  #161-D: unrecognized #pragma
[Warning] core_cm4.h@28,0:  #161-D: unrecognized #pragma
Compile [ 47.6%]: aes.c
Compile [ 47.8%]: aesni.c
Compile [ 48.1%]: utest_types.cpp
[Warning] core_cm4.h@28,0:  #161-D: unrecognized #pragma
[Warning] core_cm4.h@28,0:  #161-D: unrecognized #pragma
Compile [ 48.3%]: arc4.c
Compile [ 48.6%]: asn1parse.c
Compile [ 48.8%]: base64.c
Compile [ 49.0%]: asn1write.c
Compile [ 49.3%]: bignum.c
[Error] bignum.c@1140,0:  #18: expected a ")"
[Error] bignum.c@1151,0:  #18: expected a ")"
[Error] bignum.c@1158,0:  #18: expected a ")"
[ERROR] "..\mbed-os\features\mbedtls\src\bignum.c", line 1140: Error:  #18: expected a ")"
"..\mbed-os\features\mbedtls\src\bignum.c", line 1151: Error:  #18: expected a ")"
"..\mbed-os\features\mbedtls\src\bignum.c", line 1158: Error:  #18: expected a ")"
..\mbed-os\features\mbedtls\src\bignum.c: 0 warnings, 3 errors

[mbed] ERROR: "C:\Python27\python.exe" returned error code 1.
[mbed] ERROR: Command "C:\Python27\python.exe -u C:\mbed-os-example-client\mbed-os\tools\make.py -t ARM -m K64F --source .. --build ..\BUILD\
K64F\ARM -c" in "C:\mbed-os-example-client\configs"
---
PS C:\mbed-os-example-client\configs>

teetak01, 2017-09-19:
Hi @vijhar01, please move this issue to Mbed OS as the error originates from Mbed OS component.

ashok-rao, 2017-09-19:
@adbridge .. FYI..

teetak01, 2017-09-19:
Also, you should highlight that this was tested with mbed OS 5.6. It was not immediately appearant.

teetak01, 2017-09-19:
Application compiles fine for mbed OS 5.5.7 with GCC, ARM and IAR, so not an application issue.

ashok-rao, 2017-09-19:
@vijhar01 .. I can't seem to reproduce this.. following the exact same steps you mentioned, the application builds fine on my environment.
```
Total Static RAM memory (data + bss): 75477 bytes
Total Flash memory (text + data): 277513 bytes

Image: ..\BUILD\K64F\ARM\mbed-os-example-client.bin

C:\Ashok\mbed_CLI\OoB_5_6\Client\mbed-os-example-client\configs>
```
@adbridge @teetak01 @MarceloSalazar

teetak01, 2017-09-19:
did you do some local changes to mbedtls config? That error seems to point to MbedTLS.

teetak01, 2017-09-19:
Not a bug.

MarceloSalazar, 2017-09-19:
@vijhar01 are you still seeing this problem?
If so, could you please confirm your toolchain version?

teetak01, 2017-10-16:
@vijhar01 is this still an issue?

teetak01, 2017-10-25:
Closing due to inactivity.

closed 2017-10-25 by teetak01