ARMmbed/mbed-os-example-client #322: [ERR ][mClt]: M2MConnectionSecurityPimpl::timer_expired - handshake timeout

haishengliu, 2017-10-09:
Hello, Guys,

### Description
- Type: Bug | Enhancement | Question
[ERR ][mClt]: M2MConnectionSecurityPimpl::timer_expired - handshake timeout

---------------------------------------------------------------
## Bug

**Version information**
(`mbed ls`)

mbed-os-example-client.bak (8cf702c32443)
|- easy-connect (bf821b0695cc)
|  |- atmel-rf-driver (3cc6010188c0)
|  |- esp8266-driver (fdb44a4bedbb)
|  |  `- ESP8266/ATParser (a2cb09990f11)
|  |- mcr20a-rf-driver (9aac10474702)
|  `- stm-spirit1-rf-driver (bee61bbe6807)
|- mbed-client (31e5ce203cc0)
|  |- mbed-client-c (ecfa619e42b2)
|  |- mbed-client-classic (4e66929607c3)
|  `- mbed-client-mbed-tls (7e1b6d815038)
|- mbed-os (5499db1e815b)
`- pal (60ce64d5ec35)

**Environment details**
Hw: 
border router: Atmel shield 2.4G
client: K64F + Atmel shield 2.4G

**Expected Behavior**
registered object successfully

**Actual Behavior**
[ERR ][mClt]: M2MConnectionSecurityPimpl::timer_expired - handshake timeout
**Steps to Reproduce**
as this project indicates............
-----------------------------------------------------------------

## Question

**How to?**
How to get a work around to avoid this issue
It seems that it was working with previous release of mbed os very long time ago.  But now, this error message is very common.

Google doesn't give me lots of information. 
Any help is highly  appreciated. Thank you very much
[putty.log](https://github.com/ARMmbed/mbed-os-example-client/files/1368686/putty.log)

For detailed log info, please refer to the attached log file. 
Haisheng

teetak01, 2017-10-09:
Hi @haishengliu you could try to increase the SSL timeout parameter here:

https://github.com/ARMmbed/mbed-client-mbedtls-mirror/blob/1d7d223267f98c578fed190093a46c51284641ff/source/m2mconnectionsecuritypimpl.cpp#L274

The default maximum might be a bit too short for Mesh. For. ex. 29 => 60 or 80.

haishengliu, 2017-10-10:
Hi, @teetak01 , thank you for your advice. I do change the 29 -> 80, as  mbedtls_ssl_conf_handshake_timeout( &_conf, 10000, 80000 ); It is just not working either. 
By the way, the link above seems broken. It shows me a 404 page. 
Could you please give me any hint? Thank you in advance

Haisheng

teetak01, 2017-10-10:
Fixed the link.

Can you try if TCP works better?

Also the lifetime could be much larger. With TCP, 3600 seconds should be quite fine.

haishengliu, 2017-10-10:
I change the file simpleclient.h to 
//Select binding mode: UDP or TCP -- note - Mesh networking is *****IPv6 UDP ONLY****
#ifdef MESH
    M2MInterface::BindingMode SOCKET_MODE = M2MInterface::TCP;

I get this error message when running: 
[ERR ][mClt]: pal_connect(): failed: -65536

[DBG ][mClt]: resolve_server_address()

[DBG ][mClt]: send_dns_event()

[DBG ][mClt]: M2MConnectionHandlerPimpl::dns_ha[90m[DBG ][mClt]: M2MConnectionHandlerPimpl::dns_handler - _sondler - _socket_state = 0

[DBG ][mClt]: close_socket() - IN

[DBG ][mClt]: close_socket() - status: 0 OUT

[DBG ][mClt]: address family: 10

[DBG ][mClt]: IPv6 Address 26:07:f0:d0:26:01:00:52:00:00:00:00:00:00:00:20

[DBG ][mClt]: init_socket - IN

[DBG ][mClt]: Interface count: 1

[DBG ][nslp]: getOwnIpAddress()

[DBG ][mClt]: Interface name:

[DBG ][mClt]: Interface no: 0

[DBG ][mClt]: init_socket - port 0

[DBG ][sck ]: Socket id 1 allocated

[DBG ][nsif]: socket_open() socket=0x2000b8dc, sock_id=1, ret=0

[DBG ][nsif]: socket_attach(socket=0x2000b8dc) sock_id=1

[DBG ][nsif]: socket_bind(socket=0x2000b8dc) sock_id=1, retcode=0, ret=0

[DBG ][mClt]: init_socket - OUT

[DBG ][mClt]: resolve_server_address - Using TCP

[DBG ][tcp ]: tcp_session_ptr_allocate() s=1

[DBG ][tcp ]: isn [2a01:e35:2f47:b380:fec2:3d00:3:cbcd]:49406 -> [2607:f0d0:2601:52::20]:5684

[DBG ][tcp ]: secret init

[DBG ][tcp ]: ISN=69e4b2bf M=33e240 F=69b0d07f fudge=100

[DBG ][tcp ]: tcp_session_open()

[DBG ][nsif]: socket_connect(socket=0x2000b8dc) sock_id=1, ret=-3013

[ERR ][mClt]: pal_connect(): failed: -65536

[DBG ][mClt]: close_socket() - IN

[DBG ][nsif]: socket_attach(socket=0x2000b8dc) sock_id=1

[DBG ][nsif]: socket_close(socket=0x2000b8dc) sock_id=1

[DBG ][tcp ]: tcp_session_shutdown_read

[DBG ][tcp ]: tcp_session_close

[DBG ][tcp ]: tcp_session_delete_with_error(), s=-1

[DBG ][tcp ]: tcp_session_ptr_free() s=-1

[DBG ][mClt]: close_socket() - status: 0 OUT

[DBG ][mClt]: M2MInterfaceImpl::socket_error: (4), retry (1), reconnecting (0), reconnection_state (1)

[DBG ][mClt]: M2MInterfaceImpl::internal_event : new state 0

[DBG ][mClt]: M2MInterfaceImpl::state_engine

[DBG ][mClt]: M2MInterfaceImpl::state_idle

[DBG ][mClt]: M2MNsdlInterface::stop_timers()

[DBG ][mClt]: stop_listening()

[DBG ][mClt]: close_socket() - IN

[DBG ][mClt]: close_socket() - status: 0 OUT

[DBG ][mClt]: M2MInterfaceImpl::socket_error - reconnecting in 15(s), count 1/3

It is freezed at this stage, never restarting. I noticed that in the comments:  Mesh networking is IPv6 UDP ONLY, maybe mesh only supports UDP? get completely no idea
Thanks

teetak01, 2017-10-10:
It could be that this application does not support Mesh with TCP.

teetak01, 2017-10-13:
@haishengliu do you know how stable the IPv6 connection is? Can you try connecting with ethernet LWIP6 mode? There really does not seem to be any functional problem, other than establishing the secure connection keeps failing due to packet loss or other network-releted issues.

haishengliu, 2017-10-13:
hi, @teetak01, it seems that this is something network related. Sometimes it is working with the same setting. And the object can be connected and registered to the mbed connector. But also can be failed as discussed on this issue. I don't have notation about the ipv6 connection state. Thank you very much for your nice support.

closed 2017-10-13 by haishengliu