ARMmbed/mbed-os-example-client #351: mbed connector - backward compatibility problem?

rspelta, 2017-12-12:
- Type: Bug?

cc @MarceloSalazar 

Hello, I'm having a problem with the mbed connector website. Our [mbed-os-sensor-node](https://os.mbed.com/teams/Avnet-Silica/code/mbed-os-sensor-node/) project is not working anymore. This project is a porting from a old version of the mbed-os-example-client. The mesh works but it is not possible connect any clients to the mbed connector.
Some month ago I was able to connect 20 clients contemporaneally without problems, we have done a lot of seminars around the europe with it. Now it is impossible. My firmware is not changed at all.

mbed-os-sensor-node (0bffa083f580)
|- easy-connect (4e34947657b2) **changed for my board**
|  |- atmel-rf-driver (0ae76ce17ae5)
|  |- esp8266-driver (ebc1fbd0b53a)
|  |  `- ESP8266\ATParser (77734ee44a63)
|  |- mcr20a-rf-driver (d5905fefa54c)
|  |- sensornode-rf-driver (d836be1011b8)
|  `- stm-spirit1-rf-driver (cb5f1e808ee9)
|- mbed-client (31e5ce203cc0)
|  |- mbed-client-c (ecfa619e42b2)
|  |- mbed-client-classic (4e66929607c3)
|  `- mbed-client-mbed-tls (7e1b6d815038)
|- mbed-os (7049228c65bf) **changed for my board**
`- pal (60ce64d5ec35)

The log from the client is this:

**SOCKET_MODE : UDP
Connecting to coaps://[2607:f0d0:2601:52::20]:5684
simulate button_click, device not registered
simulate button_click, device not registered
simulate button_click, device not registered
simulate button_click, device not registered
simulate button_click, device not registered
simulate button_click, device not registered
[ERROR:] M2MInterface::SecureConnectionFailed
simulate button_click, device not registered
simulate button_click, device not registered**

If I download the last mbed-os-example-client and I change it for my board then the client connects to the mbed connector. But, the connection is not stable. It is impossible connect more than 4 clients contemporaneally.

So my starting question is, do you have changed the mbed connector and now is not compatible with the old versions of the mbed-os-example-client?

Next year we are going to do more seminars, so it is very important for us come to a solution with 20 clients connected without any problem.

Thank you!

teetak01, 2017-12-15:
@rspelta Were you trying to use old sertificate files/binaries in your initial test? The security.h files expired at some point in the beginning of this year. Other than that there has been no changes.

Could you update to the latest atmel driver + latest border-router with latest atmel driver (https://github.com/ARMmbed/nanostack-border-router). The driver has significant fixes for the packet-loss issues present in the older drivers, You need the fixed driver for both the client and router-side.

rspelta, 2017-12-16:
Hi,

> Were you trying to use old sertificate files/binaries in your initial test? The security.h files expired at some point in the beginning of this year.

I used a new certificate but still it not works, the sensor node doesn't connect to the mbed connector.

I give you the following repositories in order to replicate my issue:

* If you want to use the repository used for the seminars:
[https://os.mbed.com/teams/Avnet-Silica/code/mbed-os-sensor-node/](https://os.mbed.com/teams/Avnet-Silica/code/mbed-os-sensor-node/)
about 20 days ago I commited just to have the possibility to compile it with uvision 5. If you want to use exactly the version used in the seminars you have to dowload this commit:
[https://os.mbed.com/teams/Avnet-Silica/code/mbed-os-sensor-node/rev/3c05759bc6c8/](https://os.mbed.com/teams/Avnet-Silica/code/mbed-os-sensor-node/rev/3c05759bc6c8/)

* I forked your lastest mbed-os and mbed-os-example-client and ported them for the sensor-node:
[https://github.com/rspelta/mbed-os-example-client](https://github.com/rspelta/mbed-os-example-client)
You will see that I have my version of mbed-os, but, I just added my .ld file in order to manage correctly the RAM2 and RAM1.
With this version you should be able to connect to the mbed connector but if you use more sensor-node (using obviusly different macs and certificates) they won't connect or after a while, they will disappear from the list of the device connected.

* For the edge-router [(ST-Nucleo-F429ZI](https://os.mbed.com/platforms/ST-Nucleo-F429ZI/) + [x-nucleo-ids01a4),](http://www.st.com/en/ecosystems/x-nucleo-ids01a4.html) I give you the firmware used in the seminars:
[http://sensor-node.readthedocs.io/en/latest/_static/nanostack-border-router.bin](http://sensor-node.readthedocs.io/en/latest/_static/nanostack-border-router.bin)

* If you want to use the lastest version, I forked the nanostack-boarder-router and I modified it in order to have the lastest drivers versions and I changed the json file. You can see the commits to understand what I have changed.
[https://github.com/rspelta/nanostack-border-router](https://github.com/rspelta/nanostack-border-router)

Here there is our documentation about this project and how to setup the hardware:
[https://os.mbed.com/teams/Avnet-Silica/code/mbed-os-sensor-node/wiki/Configure](https://os.mbed.com/teams/Avnet-Silica/code/mbed-os-sensor-node/wiki/Configure)

We are using a tunnel 6to4 in order to communicate with the mbed connector.

JanneKiiskila, 2017-12-21:
That is a very long list of changes needed - are you able to any way isolate the changes one by one, i.e. change only ONE of those components to isolate which one exactly is now regressing? 

Also, we would need the more detailed logs from the client side.

MarceloSalazar, 2018-02-02:
@aratti has confirmed that the application is much more stable when switching to UDP, instead of TCP (I'd expect to be the other way round).

@SeppoTakalo would you be able to comment how is this possible?
Could you please look into this? Thanks!

rspelta, 2018-02-05:
> @SeppoTakalo would you be able to comment how is this possible?
Could you please look into this? Thanks!

The UDP/TCP problem was with a newer version of the mbed-os and that problem was an our software-compiling issue. 6lowpan mesh must work with UDP not TCP.

update:
At the moment our "seminar version" [(mbed-os-sensor-node)](https://os.mbed.com/teams/Avnet-Silica/code/mbed-os-sensor-node/) is working well. This version will be used at our seminars, and for now, we don't change it with a newer version of the mbed-os.

We don't know why for a lapse of time that version didn't work at all. The clues make me think about some problem with the ip6to4 tunneling. If this problem will repeat this will be the first thing I will delve into.
We could close this ticket.

JanneKiiskila, 2018-02-05:
Which tunnel are you using? We had some similar experiences in UK side as well, @mikaleppanen  was studying those and there was some clear tunnel issue.

rspelta, 2018-02-09:
I have 2 ways:
1. Enabling the "tunnel6to4" option by our company router connected to the internet. It is done automatically and I don't know exactly how it works.
2. Using our server over internet that implement a tunnel6to4 service.

MarceloSalazar, 2018-02-14:
@rspelta thanks for letting us know. I'm closing this ticket as the problem is gone.

closed 2018-02-14 by MarceloSalazar