ARMmbed/mbed-os-example-client #353: Wif idw01m1_v4.json

MarceloSalazar, 2017-12-13:
Note: This is just a template, so feel free to use/remove the unnecessary things

Please note that if your issue is reproducible on a specific configuration only,
it's likely an mbed-OS issue rather than mbed-os-example-client issue. We recommend
reporting the the issue then directly to 
[mbed-os reposititory](https://github.com/ARMmbed/mbed-os/issues) - please also
try searching for duplicates before filing a new one.

### Description
- Type: Bug | Enhancement | Question
- Related Issue: `#abc`

---------------------------------------------------------------
## Bug

**Version information**
(`mbed ls`)

**Environment details**
Describe against which environment you are testing.
Compiler & other relevant tools - version information.

**Expected Behavior**

**Actual Behavior**

**Steps to Reproduce**

----------------------------------------------------------------
## Enhancement

**Reason to enhance/problem with existing solution**

**Suggested enhancement**

**Pros**

**Cons**

-----------------------------------------------------------------

## Question

**How to?**

betzw, 2017-12-13:
Well, it are not the WiFi shields which are not supporting TLS/entropy but the NUCLEO base board(s) (e.g. `NUCLEO_F401RE`) which I use for testing that do not support this security level.

MarceloSalazar, 2017-12-13:
Perhaps we can switch to `NUCLEO_L4x6RG`?
I don't think `NUCLEO_F401RE` meets the requirements and should be dropped from this example app.

JanneKiiskila, 2017-12-13:
That sounds like the right conclusion. Which other base board would do?

JanneKiiskila, 2017-12-13:
(Or implement the entropy parts to that board as well, that's another way out of it).

JanneKiiskila, 2017-12-13:
There's a test matrix; https://os.mbed.com/teams/ST/wiki/Matrix-of-tested-boards
I think we could pick up another board from there?  NUCLEO L476RG potentially?

betzw, 2017-12-14:
> The ST Wifi configuration for idw01 and idw04 are using:
>
>   - MBEDTLS_NO_DEFAULT_ENTROPY_SOURCES
>   - MBEDTLS_TEST_NULL_ENTROPY

Would it help if we remove those `macros` sections of
- [`configs/wifi_idw01m1_v4.json`](https://github.com/ARMmbed/mbed-os-example-client/blob/master/configs/wifi_idw01m1_v4.json#L24), and
- [`configs/wifi_idw04a1_v4.json`](https://github.com/ARMmbed/mbed-os-example-client/blob/master/configs/wifi_idw04a1_v4.json#L24)

?!?

JanneKiiskila, 2017-12-14:
It will compile then at all, I'm afraid.

```
[Error] config.h@2743,2: #error "No entropy source was found at build time, so TLS " "functionality is not available"
[ERROR] In file included from ./mbed-client/mbed-client-mbed-tls/mbed-client-mbedtls/m2mconnectionsecuritypimpl.h:25:0,
                 from ./mbed-client/mbed-client-mbed-tls/source/m2mconnectionsecurity.cpp:22:
./mbed-os/features/mbedtls/inc/mbedtls/config.h:2743:2: error: #error "No entropy source was found at build time, so TLS " "functionality is not available"
 #error "No entropy source was found at build time, so TLS " \
```

betzw, 2017-12-14:
Which base board are you using?

JanneKiiskila, 2017-12-14:
This compilation with with using the `configs/wifi_idw01m1_v4.json` and just `mbed compile -m NUCLEO_F401RE -t GCC_ARM`.

betzw, 2017-12-14:
Well, the `NUCLEO_F401RE` does not provide support for `TLS/entropy` and such the compilation correctly produces the above errors.
You might want to try to compile for base boards which support TLS, e.g. NUCLEO boards `NUCLEO_L4x6RG`.

betzw, 2018-02-19:
@MarceloSalazar, @JanneKiiskila 
Can we close this issue?

JanneKiiskila, 2018-02-19:
Yep!

closed 2018-02-19 by JanneKiiskila